TcMenu/tcMenu #10: Make a python version of the remote control API that works over rs232

davetcc, 2018-06-05:
As it says on the tin, a Python API that will connect to and remotely control the menu library, just like it's Java counterpart

davetcc, 2018-06-05:
Issues to be thought about, how to deal with the connection loops that are threads in Java. What test application should be used etc..

davetcc, 2019-02-24:
If any python user of the library wants to have a go at implementing the protocol for python, I'd be very glad to help you understand the protocol and help out with Serial and Socket understanding..

vzahradnik, 2022-09-28:
I will need to build a library for Dart (because of [Flutter](https://flutter.dev/)) and some time later also for Python (because I will most likely build a [Home Assistant](https://github.com/home-assistant) integration). So perhaps I could work on both.

Is the remote control API identical when using RS232 vs TagValue? I will need TagValue support but ideally I could support both.

davetcc, 2022-09-28:
That would be great, I would provide as much help as I possibly could with that as it would be generally useful to many, there are four API implementations at the moment that you can look at:

* Typescript (Javascript):  https://github.com/davetcc/embedcontrolJS
* Java: TcMenuJavaAPI in this project.
* C++: In the tcMenuLib repo
* .NET: I even have an experimental C# version that was never released as isn't really complete.

All of the implementations use the same protocol and messaging, ethernet, RS232 and even the websocket version is just a thin websocket wrapper around the same protocol. It is best to assume that other protocols could be added later, and only process messages that have protocol 1 (TagVal). Probably the best starting point is: https://www.thecoderscorner.com/products/apps/embed-control/embed-control-tagval-wire-protocol/ and then take a look through the current implementations.

I'd recommend maybe starting with some unit tests and a protocol converter in one of the languages that can serialize the messages. If you create a repo for it, or if you prefer to put it here I'll do my best to help out with it as well.

vzahradnik, 2022-09-29:
I think the easiest approach will be if I create the repos and later I will transfer ownership to you. From the start, there will be a lot of commits and I don't want to waste anyone's time on approving the work-in-progress code.

Allright, I will start by looking at the existing code and we can discuss in this thread. Later, if it will be necessary, we can also chat through some other clients.

Thanks!

davetcc, 2022-11-05:
How is it going on the library front, feel free to add me on any repo you create, I've done a few different ports of the API now, and I could probably help get something workable quite quickly. You can ping me using the contact form on thecoderscorner or PM me in the forum.

vzahradnik, 2022-11-05:
Hi @davetcc I had busy month working on non-coding stuff. I have this in my mind and will definitely ask you when I hit on some block.

vzahradnik, 2022-11-12:
@davetcc what do you use for testing besides writing unit tests for the parser? I will need to test realtime communication between the library and the embedded server. Ideally I would use something like an embedded simulator on my local PC if there's anything like that.

davetcc, 2022-11-14:
Let me know if you have trouble building it and I’ll jpackage it to somewhere for now.

davetcc, 2022-12-01:
It struck me recently that if all you want is a UI for iOS and Android, another starting point may be embedCONTROL.js

With a React.native front end, you could target both mobile devices very easily and the API is pretty much already written and working, at least for WebSockets. A full socket and Bluetooth implementation could follow for the native version.

There is already a web front end working that works well with mobile, If you run the embeddedJavaExample in the tcMenu project the web UI is served from port 8080, if native bindings were used instead of webpage bindings it may do most of what you need.

https://github.com/davetcc/embedcontrolJS

vzahradnik, 2022-12-03:
Hi Dave,

I came to a similar idea to you. Just instead of embedCONTROL.js I would use Flutter. There is a library to embed JS code.
Anyway, I still think native versions would be better. Eventually, I will most likely need Python library too.

I already started my work on the app, except it is not bound to actual data. Now I start in paralell working on the library. My guess is that it won't be that complex once I get the gist of it.

https://www.youtube.com/watch?v=T_nFXBiCyZ4

By the way, does this protocol support embedding custom data/messages? Because I'd like to send additional information, like status messages. I have a status screen and I want users to know just by looking at the display what is the state of the device.

Thanks!

vzahradnik, 2022-12-04:
@davetcc 

From your description I see the implementations vary in supported functionality. In ideal case, what should the library support?

I came to this so far:
- TagVal over serial connection, e.g., BlueTooth
- TagVal over IP
- TagVal over WebSockets

It looks like the JS version supports just the WebSockets, while Java supports IP and serial connections, right?

How do you implement WebSockets on Arduino side?

Thanks!

davetcc, 2022-12-06:
It's by far safe to say that the Java API is the most complete and most tested, if you were copying either into Dart, I would use the Java one, as you'll come out with more or less everything (WebSocket support, IP support, Serial support).

The JS implementation at the moment only supports running in a browser, so only supports web sockets as that is all it can support in that context. However, I was thinking that a react native UI may not be that difficult from where it is. At the moment, the websocket support requires a Java server to run that talks the regular protocol to Arduino and runs a jetty web server. It was originally envisaged to run on a RPI or similar, and probably run as if it were a regular IoT component. See the menu in menu docs on the main site if you wanted more background on that.

I tried an Arduino implementation that could serve up a react app and then run a websocket, but the HTTP requests needed were just too large, initially running into problems with AsyncWebServer, even trying writing my own, and whatever way required amounts of memory that were demanding even for ESP32 or STM32. I decided to park that idea for now, but you can see it's all checked into another experiment repo as it were. https://github.com/davetcc/TcMenuNetLayer

vzahradnik, 2022-12-06:
Oh, it makes sense. So the WebSockets are just the bridge to the Java API server.

I plan to build my version on the Java one. From what I can already tell:
- I need to create data model
- I need to implement menu manager
- I need to implement TagVal

Having all those pieces, the Python/Dart libraries could support both client and server. At this point it probably won't make much sense as we're talking to Arduino/RPi, but perhaps there will be a usecase in the future. Also, it would help with testing having all bits and pieces together. I for sure could use a Python server. Something serving a simple menu with two entries.

Python library could optionally support websockets as well. There is a library to provide support. But I'll start with what I need the most and then we'll go from there - a client supporting IP connections.

I will keep updates in this thread. And once there will be something useful, we can discuss transfering the repository under you.

vzahradnik, 2022-12-06:
By the way, didn't you consider separating the API part from the codebase? Like a separate repository called for example `tcmenu-api-java`. Especially when there will be more libraries, it makes sense, at least to me.

davetcc, 2022-12-12:
It kind of is still separate, even though it's in the main package at the moment, it builds into maven central just as the API: https://search.maven.org/search?q=g:com.thecoderscorner.tcmenu

There are pluses and minuses to separating it, but the dart and other implementations would need their own repo anyway.

Probably the best thing to do is drop me an email (from my profile there's an email option) and we can maybe catch up on the best way forward, for both this and i18n support.

vzahradnik, 2022-12-12:
Sounds good. By the way, I have almost complete data model ready. MenuTree is TBD.

https://github.com/Lutemi/tcmenu-core-python/tree/main/tcmenu/domain/state

davetcc, 2023-01-13:
Should we move the conversation into discussions on this repo for now? Discord doesn't send notifications well, and anything relatively private that is not suited to the discussion, just email me.