TcMenu/tcMenu #35: Monochrome Adafruit GFX driver for displays such as OLED 1306 and Nokia 5110

davetcc, 2019-03-03:
There needs to be a display driver for Adafruit GFX that is optimized for monochrome displays such as a 1306 OLED unit and the Nokia 5110 display.

micahvdm, 2019-04-08:
I'm struggling to get my menu to compile with SSD1306. Seems to give errors about the renderer.

davetcc, 2019-04-08:
Hi there many thanks for raising the issue. In V1.3 and then V1.3.1 we are focusing on major improvements to the ada graphics driver.

V1.3 release will be coming in the next week and you should find it much easier to use with a wider range of such displays.

If you could provide more detail as to the error it would be helpful. Please include the version of TcMenu you’ve used, a sketch that recreates it and the errors you see. 

Feel free to privately post source using the contact form on thecoderscorner (linked on the main page) should you prefer not to submit here.

micahvdm, 2019-04-09:
Many thanks for your reply. I will recreate an example when I get home from work and send it to you.

micahvdm, 2019-04-10:
[test.zip](https://github.com/davetcc/tcMenu/files/3065265/test.zip)
Here's the test menu with SSD1306 and log

davetcc, 2019-04-11:
Problem located, in the attached sketch: test.ino, you need to create a display variable. For example to use this constructor

```
 Adafruit_SSD1306(uint8_t w, uint8_t h, TwoWire *twi=&Wire, int8_t rst_pin=-1,
    uint32_t clkDuring=400000UL, uint32_t clkAfter=100000UL);
```

You may add something similar to the following to your sketch, using *exactly* the same name as you used in the menu designer UI for the graphics variable:

```
#include "test_menu.h"
// Note that you must create the actual variable. 
// There are just too many combinations for the menu designer UI to create the variable
// The menu UI just creates an external reference to this variable.
Adafruit_SSD1306 gfx(128, 32);

void setup() {
    setupMenu();

}

void loop() {
    taskManager.runLoop();

}
```

When I added this your sketch compiles. I did not have a 1306 around to test with. There are a few changes around this in V1.3 and V1.3.1 that will make it easier to customise rendering. I will also be adding an example to the list of examples for this too.

davetcc, 2019-04-11:
Just be aware support for these monochrome displays is not yet tested. I cannot say it will renderer well with anything less than V1.3 of the embedded library; which is due to be released shortly. The AdaGFX support is currently only tested with colour TFTs.

Once 1.3 is released, the API will be stable, setting up rendering colours and spacing will be possible, and a considerable amount of additional documentation will follow. Unless you're prepared to take master, I'd wait for 1.3 to be released.

micahvdm, 2019-04-11:
Thanks for the reply. It unfortunately doesn't render anything on the display. seems like there will be more to it that just initiating the display.

davetcc, 2019-04-11:
Yep, until V1.3 it will try to render in colour, and I assume that won’t
work. With the new version you’ll be able to provide a graphics
configuration and set the colours to black and white.

I have a couple of monochrome displays to test with but they need soldering
up. I’ll try one with the master code and feed back with the results.

Thanks
Dave.

davetcc, 2019-04-13:
Good news is I’ve worked out why it did not render and should have a fix that will be tested with colour and 5110 by tomorrow. There is no reason it shouldn’t also work with the 1306 unit too.

davetcc, 2019-04-19:
Working on master, will be release later today in 1.3

davetcc, 2019-05-04:
added additional debugging - OLED 1306 not working properly.

davetcc, 2019-05-05:
Improvements for both Nokia and OLED units.

sebacavadini, 2019-11-20:
Hi all, 
I am using TcMenu with Arduino Mega 2560 and 128x64 NHD HW SPI.
I was able to compile and run with the U8g2 library but there is a lot of problems with the rendering,
I`m trying to make it run with TcMenu with Adafruit GFX library and I do not know whats going on.
This is my configuration for generate code:
[image]

This is my .ino file:
`#include "Arduino_menu.h"

Adafruit_SSD1325 display(9, 8, 10);
    
void setup() {
    display.begin();

    setupMenu();

}

void loop() {
    taskManager.runLoop();

}
`

And I get many compilation errors like: C:\Users\User\Documents\Arduino\libraries\Adafruit_SSD1325/Adafruit_SSD1325.h:104:7: note: previous definition of 'class Adafruit_SSD1325'

 class Adafruit_SSD1325 : public Adafruit_GFX {

       ^~~~~~~~~~~~~~~~

Multiple libraries....

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Do you have an idea what i am doing wrong?

Thanks for your help.

davetcc, 2019-11-20:
> U8g2 library but there is a lot of problems with the rendering,

Please could you attach an image showing the u8g2 problems. I test with a 1306 driver frequently but not the one you are using. 

> adafruit 1325 library:

this library is missing multiple inclusion defends around the include file. I’ve got a task from a while ago to raise a PR against this library. For now you could add at the top of the header file:

```
#ifndef ADA1325LIBRARY
#define ADA1325LIBRARY
```

And then at the end of the header file 

`#endif`

davetcc, 2019-11-21:
I just took a look over at the adagfx 1325 repo and there is already a pull request (https://github.com/adafruit/Adafruit_SSD1325_Library/pull/1) to fix the include guard, but it's not been merged in some time.

Given this report has cropped up several times in emails, forum posts and here in this issue, if it doesn't get merged at source, I'll have to try and do something in the designer to prevent the issue..

davetcc, 2020-07-18:
Closing this and opening a new issue to track the problem with the 1325 repo

closed 2020-07-18 by davetcc