TcMenu/tcMenu #52: Adding Numeric Keypad

Prithul0218, 2019-07-15:
Is it possible to add numeric keypads like these to the library? [https://cdn.instructables.com/FSM/WKU0/HX0DQ2Y3/FSMWKU0HX0DQ2Y3.LARGE.jpg](url)

It will be really useful to take values from the user. Also having to option to use this keypad with a rotary encoder will be really cool. One can, say, make a digital power supply with it.

Thank you

davetcc, 2019-08-09:
Support for matrix style (8 pin) units will be added in 1.3.3; which should be released in about a week.

Prithul0218, 2019-08-09:
Thank you so much. I will try to test it for you.

davetcc, 2019-08-11:
This is in progress but not yet ready for testing. Majority of the work is in the library repo under the following issue: https://github.com/davetcc/tcMenuLib/issues/34

**For 1.3.3 the following will be targetted for editing with the keyboard:**

* AnalogMenuItem editing (for at least most decimal powers of 10 and half increments).
* EnumMenuItem editing (where there are less than 10 items)
* BooleanMenuItem buy toggling the activate button.

**Followed by 1.3.4 support will be added for:**

* Any remaining work left in Analog item
* Ip Address menu item
* Time menu items
* Text menu items - less useful unless the matrix includes all keys

**Expected features in additional releases:**

* Pin Security dialog when it first appears will support keyboard input.
* Ability to use instead of a rotary encoder (for up / down / select actions).

davetcc, 2019-08-18:
In this release, all editable items should work out of the box. This feature is considered early access so the designer won't yet build the code, instead see the new keyboard example in tcMenu where the keyboard objects are well documented, also see https://www.thecoderscorner.com/products/arduino-libraries/io-abstraction/matrix-keyboard-keypad-manager/. 

There may be some glitches in the interaction between rotary encoder and keyboard, please report any you find in a bug report.

closed 2019-08-18 by davetcc