TcMenu/tcMenu #53: Generated Menu item names should include submenu

davetcc, 2019-08-07:
When naming a menu variable it should include the submenu for example

If the submenu were named settings and the item named contrast:

menuSettingsContrast. 

To be determined: should only the immediate submenu be included in this or every submenu back to root? Doing recursive back to root would end up with 

menuSettingsAdvancedItemName. 

One option would be a setting to control this that had three options:

* Do not include submenu name
* Include only nearest submenu name
* recursively include all submenu names.

millab, 2019-08-09:
A setting to choose naming convention would be great, otherwise I think defaulting to a full tree naming structure would be the most rigid. 

Other options would be:

- Detection of duplicates which would trigger a unique identifier for the end of the name

- Display the generated name and allow it to be editable

millab, 2019-08-09:
I'm not sure if there are restrictions on variable name length in C++ that would need to be considered?

davetcc, 2019-08-09:
I think the maximum length of a variable is quite long in GCC although I’ll try some really long ones to see. As it’s a quite small fix it can go with a couple of other small fixes into 1.3.3, the next release.

closed 2019-08-11 by davetcc

davetcc, 2019-08-14:
So to conclude on this. The maximum variable length is 64 chars, which should fit all but the longest of menu structures. 

There will be a checkbox on the code generator page, to turn on and off fully resolved names. New projects will default to ON. Existing projects will default to Off - but can be switched on as needed.