TcMenu/tcMenu #56: Request: Support for the nrf52840 Nano 33 BLE and Sense

LittleHobbyShop, 2019-08-22:
Please add support for the new Nano 33 mbed boards

davetcc, 2019-08-27:
I’m glad to be able to say that mbed is the next target for TcMenu. 

The plan of action is to get ioabstraction [davetcc/ioabstraction] working first on mbed and then immediately follow that with full designer support in TcMenu. 

Phase 1 will be ioabstraction supports mbed
Phase 2 will be TcMenu able to compile against arm
Phase 3 will be to fully support from the designer will suitable display plugins.

davetcc, 2019-08-28:
See the prerequisite task:
https://github.com/davetcc/IoAbstraction/issues/64

davetcc, 2020-05-14:
I know this was raised an absolute age ago, but we are finally doing this work now, take a look at the IoAbstraction issue link below and you'll see there's some pretty heavy lifting going on in IoAbstraction:

https://github.com/davetcc/IoAbstraction/issues/64

davetcc, 2020-06-13:
Linked with #86 - mbed and platform IO support, tcMenu and IoAbstraction development are now completely moved to Clion and platform IO using both mbed and Arduino chains. There's a branch with very early mbed support slowly building up there.

I know it's been a while, but something worthwhile is on its way very soon.

davetcc, 2020-06-25:
All merged back to master, some things left to iron out before going mainstream, but much further along:

1. We need to complete the mbed work in the Java UI
2. We need to re-release the windows UI to the store with mbed support.

davetcc, 2020-07-17:
The first version of this now works and is committed on tcMenuLib master (https://github.com/davetcc/tcMenuLib/issues/67), after some thought we used the Arduino abstraction, even though it's underpinnings are mbed, this gives the widest possible  compatibility for the user base. Using the new PlatformDetermination system in IoAbstraction, this could easily be changed for specific projects that wanted an mbed first approach.

davetcc, 2020-07-17:
Will be released in V1.5.0 soon. Passed all tests in development, 1.5.0 now going to be BETA tested.

closed 2020-07-17 by davetcc