TcMenu/tcMenu #58: M5Stack?

tlrobinson, 2019-09-26:
I've almost gotten tcMenu working with M5Stack, which has a ILI9341, however it's mirrored horizontally and seems to be cut off at maybe 240 pixels wide. Here's the code I'm using:

```c
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"

#include "M5Display.h"

#define BLK_PWM_CHANNEL 7

//#define TFT_DC 27
//#define TFT_CS 14
//#define TFT_MOSI 23
//#define TFT_MISO 19
#define TFT_CLK 18
//#define TFT_RST 33

Adafruit_ILI9341 gfx = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);

#include "menutest_menu.h"

void setup() {
  setupMenu();
  
  // turn on backlight
  ledcSetup(BLK_PWM_CHANNEL, 44100, 8);
  ledcAttachPin(TFT_BL, BLK_PWM_CHANNEL);
  ledcWrite(BLK_PWM_CHANNEL, 80);

  gfx.begin();
}
``

davetcc, 2019-09-26:
What are the settings for the display in your designer project?

Have you rotated the display to match the settings you’ve chosen using the rotate method?

I have an MKR board with such a display using the adafruit library. I’m not sure what the M5 display is to be honest.

davetcc, 2019-09-26:
Forgot to say in the last mail: Thanks for trying to port to M5, I’m sure we should get it working pretty quickly, as many of the components you’re using are already supported.

tlrobinson, 2019-09-26:
I've tried `gfx.setRotation(...)` with 0-7. It rotates the display but everything is still mirrored and it only uses a 240x240 portion of the screen.

I also tried just drawing some text using Adafruit_GFX directly and it has the same problem, so it seems like it's not specific to tcMenu.

I've found a few issues that sound similar but haven't been able to figure it out yet:

https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/41

https://github.com/Bodmer/TFT_eSPI/issues/96

davetcc, 2019-09-26:
If you can get the text working without TcMenu, then TcMenu should just work as it’s well tested with both ILI9341 and ESP hardware.

davetcc, 2019-10-13:
I'm going to close this for now as it's not really a tcMenu issue. I'm sure tcMenu would work with these boards if the display were working properly. If you get the display working properly outside of tcMenu and need further assistance, please feel free to re-open.

closed 2019-10-13 by davetcc