TcMenu/tcMenu #63: EnumItem in root with 3 items causes loop or crash

Peteraguy, 2019-10-12:
Arduino IDE 1.8.10, Arduino Uno, I2C 4x20 LCD, and rotary encoder. 
If an Enumeration Item is added in the root with more than 2 items the display enters an infinite loop or crashes (not sure which just flashes)

The following is project file
`{"version":"1.00","projectName":"C:\\Users\\USERNAME\\Documents\\Arduino\\testmenu\\testmenu2.emf","author":"Pete","lastEdited":{"seconds":1570912633,"nanos":156523500},"namingRecursive":false,"items":[{"parentId":0,"type":"enumItem","item":{"enumEntries":["Item1","item2","Item3"],"name":"New EnumItem","id":1,"eepromAddress":-1,"readOnly":false,"localOnly":false}}],"codeOptions":{"embeddedPlatform":"ARDUINO","lastDisplayUuid":"d4c84106-fa87-4024-b91e-053e17a52b8f","lastInputUuid":"6dd87e7d-e751-4912-a70d-91793b1d3d87","lastRemoteUuid":"2c101fec-1f7d-4ff3-8d2b-992ad41e7fcb","applicationUUID":"25c3ec06-60ea-4102-a8b7-1a4a2b1eb908","applicationName":"MenuTest","lastProperties":[{"name":"LCD_WIDTH","latestValue":"20","subsystem":"DISPLAY"},{"name":"LCD_HEIGHT","latestValue":"4","subsystem":"DISPLAY"},{"name":"I2C_ADDRESS","latestValue":"0x27","subsystem":"DISPLAY"},{"name":"PIN_LAYOUT","latestValue":"RS_RW_EN","subsystem":"DISPLAY"},{"name":"PULLUP_LOGIC","latestValue":"true","subsystem":"INPUT"},{"name":"INTERRUPT_SWITCHES","latestValue":"false","subsystem":"INPUT"},{"name":"SWITCH_IODEVICE","latestValue":"","subsystem":"INPUT"},{"name":"ENCODER_PIN_A","latestValue":"2","subsystem":"INPUT"},{"name":"ENCODER_PIN_B","latestValue":"3","subsystem":"INPUT"},{"name":"ENCODER_PIN_OK","latestValue":"4","subsystem":"INPUT"}],"namingRecursive":true}}`

davetcc, 2019-10-13:
Thanks for including the project, I'll try it today as I'm preparing a release and this should be simple to track down.

davetcc, 2019-10-13:
Hi there I've tried this on a Genuine Uno, 16x2 I2C backpack based on PCF8574, with switches for the encoders. With this configuration, the above project worked fine for me.

Here is the sketch (which other than my configuration settings is all I changed:

```
#include "tempUnoCrash_menu.h"
#include <Wire.h>

void setup() {
    Wire.begin();

    setupMenu();
}

void loop() {
    taskManager.runLoop();
}
```

Here's the EMF file:

```{"version":"1.00","projectName":"C:\\Users\\dave\\Documents\\Arduino\\tempUnoCrash\\unoCrash.emf","author":"dave","lastEdited":{"seconds":1570977735,"nanos":933755300},"namingRecursive":false,"items":[{"parentId":0,"type":"enumItem","item":{"enumEntries":["Item1","item2","Item3"],"name":"New EnumItem","id":1,"eepromAddress":-1,"readOnly":false,"localOnly":false}}],"codeOptions":{"embeddedPlatform":"ARDUINO","lastDisplayUuid":"d4c84106-fa87-4024-b91e-053e17a52b8f","lastInputUuid":"2a3ae3ba-05f4-4c9c-aad3-3ddc88df9049","lastRemoteUuid":"2c101fec-1f7d-4ff3-8d2b-992ad41e7fcb","applicationUUID":"25c3ec06-60ea-4102-a8b7-1a4a2b1eb908","applicationName":"MenuTest","lastProperties":[{"name":"LCD_WIDTH","latestValue":"20","subsystem":"DISPLAY"},{"name":"LCD_HEIGHT","latestValue":"4","subsystem":"DISPLAY"},{"name":"I2C_ADDRESS","latestValue":"0x27","subsystem":"DISPLAY"},{"name":"PIN_LAYOUT","latestValue":"RS_RW_EN","subsystem":"DISPLAY"},{"name":"PULLUP_LOGIC","latestValue":"true","subsystem":"INPUT"},{"name":"INTERRUPT_SWITCHES","latestValue":"false","subsystem":"INPUT"},{"name":"SWITCH_IODEVICE","latestValue":"","subsystem":"INPUT"},{"name":"ENCODER_UP_PIN","latestValue":"8","subsystem":"INPUT"},{"name":"ENCODER_DOWN_PIN","latestValue":"9","subsystem":"INPUT"},{"name":"ENCODER_OK_PIN","latestValue":"10","subsystem":"INPUT"}],"namingRecursive":true}}```

Is your backpack a standard i2c backpack based on 8574? Is the wiring correct and checked with a simpler program that prints in a loop to the display?

The library can deal with non-standard backpacks, but it needs more configuration.

Peteraguy, 2019-10-13:
Hi dave, thanks for getting back. I am also running this on a genuine uno and the backpack is 8574T. Wether the T makes a difference, I have not looked. The menu generator is great and I can get it to do all sorts of menus correctly on this wiring/hardware. I had a 3 item enumitem in a sub menu which worked. It was just when I tried to put one in the root I had trouble. But I probably need to sit down and do some clean tests as it was feeling messy. It put a two item in root it was fine, put three in problems, reduced it to two still problems. If it works fine for you I will swap some bits around and do some clean tests and see where I get to.
Many thanks Pete

davetcc, 2019-10-26:
Closing out for now, please re-open if the problem persists.

closed 2019-10-26 by davetcc