TcMenu/tcMenu #67: Unable to get the simpleU8g2_menu example to work on my Arduino

DeviousPenguin, 2019-11-26:
Hi, I'm using a SparkFun pro Micro board which is very similar to the Arduino Leonardo.

I'm trying to compile the simpleU8g2_menu example with a SH1106 OLED but it is saying that it the firmware is too large:

`Error: The program size (32776 bytes) is greater than maximum allowed (28672 bytes)`

So far I've only changed the basic settings in change the screen from a SSD1306 to a SH1106 and my GPIO expander to a MCP23017.

I've uploaded my sketch to a public repo here:

https://github.com/DeviousPenguin/SqueezeboxArduino

If you want to see what files I've changed, the `SqueezeboxArduino.cpp` is the file I've modified.

Are there any settings I can change to reduce the firmware size a little? It looks like it's only 4K over the limit.

Would using the Adafruit library instead work here? I've checked that the screen works OK with the U8g2 lib in an example U8g2 sketch.

davetcc, 2019-11-26:
To be honest in terms of 32K of Flash, that's kind of the minimum that the library can work with. It works OK when working with LCD screens, but I have to say that using Uno with a graphical display has not been considered this far. You could use the AVR memory tools to see what's taking up memory and see if there's anything that could reduce it.

To start with you could use: 

https://www.thecoderscorner.com/electronics/microcontrollers/efficiency/evaluating-static-memory-usage-in-arduino-sketch/

It sounds like you're using Wire, Graphic display and possibly Serial. Everything that's enabled will increase the flash size. Looking at the above map files may help.

DeviousPenguin, 2019-11-27:
Yes I am using an SPI connected display, and I2C connected GPIO expander, so this does use extra space by the looks of it, if my display was using I2C then would use a little less space. I think the best thing for me to do here would be to move over to a character LCD for this project, graphics aren't needed.

Thanks for the link about the AVR, I wanted to be sure that I didn't have some settings wrong or included lots of optional extras. I've uploaded the output from the `avr-objdump` command to a file [here](https://github.com/DeviousPenguin/SqueezeboxArduino/blob/master/avr-objdump.txt) if you want to take a look, but I really haven't added much extra that the example didn't already have. I did actually managed to just get under the memory limit by changing the font to `u8g2_font_6x10_tn`. However this font only includes numbers, no letters, so isn't too useful, I just wanted to see if I was able to go below the limit and do a quick test.

If there is anything I've missed or needlessly added then it would be nice to know, but I'll be move over to a character LCD so the limited memory should not be an issue anymore, so I'll close this issue, but feel free to follow up if there's anything I've missed that is using up uneeded memory.

closed 2019-11-27 by DeviousPenguin

davetcc, 2019-11-28:
On a device with such a small memory footprint, I would probably go with LCD too, it's what we test our Uno board with for that reason. The Uno test sketch we use with i2c, Serial and a Serial remote comes in at about 29K FLASH, 1K RAM. Even on LCD, you can have title widget graphics in the top right corner.

For title widgets see: https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/rendering-with-tcmenu-lcd-tft-oled/

There are some very nice (and therefore expensive) LCD screens that would look much nicer in a project than the regular cheap ones.

davetcc, 2019-11-28:
Just taken a quick look at the map file.

As I suspected there's Serial, SPI and Wire in there along with U8G2 and TcMenu. That's quite a lot of stuff for a board with 32K FLASH to handle!

DeviousPenguin, 2019-11-28:
> Just taken a quick look at the map file.

Thanks very much, I've ordered a simple 20x4 Character LCD with I2C connection for testing, hopefully it will be here by the weekend.

I've tried building a sketch without SPI and U8G2, and the RAM usage is much less and I'm able to fit everything with room to spare. Once I've got the basics working I'll try out the widgets feature.