TcMenu/tcMenu #68: Can not have a submenu in first position.

poorredneck, 2020-03-27:
If you have a submenu as the initial item from root position, the menu will start within the first submenu, not root/title screen. You can not exit the submenu to get back to the title screen. Submenus nestled within this first submenu do naviate normally.

I would like to not have anything on title screen other than submenus, this isn't possible at least with the way the code is generated. 

I'm just getting started with tcMenu, and am an amateur at best with arduino/C so I don't have the skill sets to come up with a solution.

davetcc, 2020-03-27:
Submenu as first item: This is actually a recent change that caused this. It is a bug and we will fix before 1.4 is released shortly. 

In terms of the LCD rendering I’ll look into feasibility of making the title configurable and comment again.

davetcc, 2020-03-27:
See 

https://github.com/davetcc/tcMenuLib/issues/48 

and 

https://github.com/davetcc/tcMenuLib/issues/47

both fixed on master, will be in 1.4 coming very soon.

closed 2020-03-27 by davetcc