TcMenu/tcMenu #73: Instructions for running the GUI under Linux

moritz89, 2020-04-20:
There have been several references to being able to run the Java GUI under Linux. Would it make sense to publish these instructions in the README?

davetcc, 2020-04-21:
If you have instructions for running on Linux it would be great to share them. I would say create a new markdown file linked from the front page. Maybe create a file called run-deisgner-on-linux.md within the tcMenuGenerator package.

moritz89, 2020-04-21:
I was hoping that you’d already have experience: “It should also be easy to run on Linux - contact me if interested.”

davetcc, 2020-04-21:
Sorry, I misunderstood, I thought you had run it on Linux and had instructions in which case I would have added them.

Although we run many server-side java processes on Linux servers, none of us have a desktop Linux environment. However, in theory, it should work as long as you have a JDK and maven available. But theory is not practical experience.

Here I openly present the current situation:

Note that the following does not include the Java API, which will continue to be developed to the same level as the C# API.

I really hate to say anything bad about Java, but JavaFX UI technology was broken out from the JDK and is now supported on Java 11 only commercially. We do not plan to move the Java designer code forward beyond Java 11. Instead, we have moved all UIs to Xamarin C# supporting Mac, Windows (and eventually iOS and Android for the remote control). These will start to be released very soon in their associated stores.

We do plan to keep the Java version going for existing users (mainly Windows 7 and 8 users) for at least another year, there will be features in the Xamarin releases that are not in the Java version, but it will continue to work for the foreseeable future.

I'll be honest, I was intending to remove the sentence about Linux at the next release (very soon).

moritz89, 2020-04-22:
That's an interesting turn of events. I also wouldn't mourn the loss of JavaFX. If the C# Xamarin code will also be released open source, I'd take a stab at running it on Linux as it should also be [supported there](https://stackoverflow.com/a/53317021/6783666).

ptapping, 2020-04-22:
On my Arch Linux machine `mvn compile` is successful, but it won't run because OpenJFX doesn't include the required javafx.application package (and maybe also javafx.scene and others). It might be possible to download the Oracle Java SDK and run it with that, but it requires registration to download, so that's as far as I got. I tend to stay away from any product with an Oracle logo on it...

davetcc, 2020-04-22:
I don't think that anything beyond the Open JDK 11 should be needed. On both Windows and MacOS I run it that way. JavaFX apps are a little different to regular JDK apps and need some additional packaging. 

Can you please try the following:

`mvn -DskipTests clean install -f pom.xml`

This will create a target directory. Within there will be a jfx/app directory.

Take a look at the win packager file at the root of the generator project to get some ideas how to run it. It's pretty well documented there.

davetcc, 2020-04-22:
>  It might be possible to download the Oracle Java SDK and run it with that, but it requires registration to download

I wouldn't attempt to download the Jdk11 from Oracle for sure, it's commerical only now, it's only got LTS from the open-source community. But if you follow the instructions above I don't think you'll need it. Certainly not needed on Windows and MacOS.

moritz89, 2020-04-22:
I've compiled the modules with openjdk/openjfx as well as Oracle Java. When testing the application, however, I run into the exception below. Any ideas?

```
java --module-path . --add-modules com.thecoderscorner.tcmenu.menuEditorUI com.thecoderscorner.menu.editorui.MenuEditorApp
Apr 22, 2020 11:25:14 PM javafx.fxml.FXMLLoader$ValueElement processValue
WARNING: Loading FXML document with JavaFX API of version 11.0.1 by JavaFX runtime of version 11
Exception in Application start method
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:1051)
Caused by: java.lang.RuntimeException: Exception in Application start method
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:900)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication$2(LauncherImpl.java:195)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.nio.file.NoSuchFileException: plugins
	at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116)
	at java.base/sun.nio.fs.UnixFileSystemProvider.newDirectoryStream(UnixFileSystemProvider.java:432)
	at java.base/java.nio.file.Files.newDirectoryStream(Files.java:471)
	at java.base/java.nio.file.Files.list(Files.java:3696)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.generator.plugin.DirectoryCodePluginManager.loadPlugins(DirectoryCodePluginManager.java:53)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.MenuEditorApp.start(MenuEditorApp.java:54)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$9(LauncherImpl.java:846)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runAndWait$12(PlatformImpl.java:455)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:428)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:427)
	at javafx.graphics/com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:277)
	... 1 more

```

ptapping, 2020-04-23:
@davetcc Thanks for advice. I haven't used Java for a while, so not familiar with JavaFX. I just saw the exception thrown from the `import javafx.application.Application` line, and that the javafx.application.jar isn't in the JavaFX libraries...

Good news is I got it to run! @moritz89 It turns out you need to manually make the `plugins` directory in the right place, which is what that exception is complaining about. Here's how I did it:

```bash
# Ensure java and maven build system installed
# Arch Linux
sudo pacman -Sy jdk11-openjdk java11-openjfx  maven
# Java version can be selected with
# sudo archlinux-java set java-11-openjdk

# Ubuntu, Debian (untested, should work)
sudo apt update && sudo apt install openjdk-11-jdk maven
# Java version can be managed with
# sudo update-alternatives --config java

# Get source code
wget https://github.com/davetcc/tcMenu/archive/1.3.5.tar.gz
tar xvf 1.3.5.tar.gz
cd tcMenu-1.3.5/tcMenuGenerator

# Build
mvn -DskipTests clean install -f pom.xml
mvn compile
mkdir target/jfx/app/plugins

# Run
cd target/jfx/app/
java --module-path ../deps --add-modules com.thecoderscorner.tcmenu.menuEditorUI com.thecoderscorner.menu.editorui.MenuEditorApp

```
Haven't actually tested it yet, but will do soon.

ptapping, 2020-04-23:
Additionally, this works to get a newer, but unreleased 1.4.0 version of the code:

```bash
# Ensure java and maven build system installed
# Arch Linux
sudo pacman -Sy jdk11-openjdk maven
# Java version can be selected with
# sudo archlinux-java set java-11-openjdk

# Ubuntu, Debian (untested, should work)
sudo apt update && sudo apt install openjdk-11-jdk maven
# Java version can be managed with
# sudo update-alternatives --config java

# Get source code for 1.4.0
# Could use --recursive to get submodules, but 1.4.0 doesn't allow installation of them anymore
git clone https://github.com/davetcc/tcMenu.git
cd tcMenu
git checkout 9c502b0
git switch -c 1.4.0

# Build plugins
mvn clean install -f baseInputDisplayPlugin
mvn clean install -f dfRobotCodePlugin

# Build app
mvn -DskipTests clean install -f tcMenuGenerator

# Run
cd tcMenuGenerator/target/jfx/app/
java --module-path ../deps --add-modules com.thecoderscorner.tcmenu.menuEditorUI com.thecoderscorner.menu.editorui.MenuEditorApp
```

davetcc, 2020-04-24:
Thanks for the instructions, I'll get these instructions moved to a more appropriate place in the repo.

davetcc, 2020-04-24:
If someone could check-out the Link from the front page in the (Installation and documentation) section and close this issue if happy with the link, otherwise let me know what else needs adding or raise a PR and I'll merge in.

Thanks again,
Dave.

moritz89, 2020-04-24:
Looks great. Wanted to ask when the C# variant of the GUI will be released?

ptapping, 2020-04-25:
Also looks fine, yeah close this for now.
I've been trying to get a universal .jar built, which would be a nicer way to run this (all dependencies included, can just java -jar ....) and be the next step to making actual Linux distribution specific packages. It's mostly working, but there's still a couple of issues with loading the plugins. If/when I get it sorted I'll make a pull request.

davetcc, 2020-04-25:
If you get the latest jpackager build and follow the Mac jpackager instructions in the windows deployment instructions it should build a native Linux package that can be installed.

ptapping, 2020-04-25:
I'm pretty sure that would work, but I believe it only supports making .deb or .rpm packages. I'm on Arch, and I don't think the jpackager binary is available. The .jar way is nice too since it doesn't need installing (and essential if you don't have root privileges).

moritz89, 2020-04-26:
I've gone through the build commands and had to adjust them slightly. I'm also running Ubuntu 20.04 on Wayland with openjdk 11 and had to set `GDK_BACKEND=x11` to avoid a crash.

```
# Ubuntu, Debian (untested, should work)
sudo apt update && sudo apt install openjdk-11-jdk maven
# Java version can be managed with
# sudo update-alternatives --config java

# Get source code for 1.4.0
# Could use --recursive to get submodules, but 1.4.0 doesn't allow installation of them anymore
git clone https://github.com/davetcc/tcMenu.git
cd tcMenu
git checkout 9c502b0
git switch -c 1.4.0

# Build plugins
(cd baseInputDisplayPlugin/ && mvn clean install -f pom.xml)
(cd dfRobotCodePlugin/ && mvn clean install -f pom.xml)

# Build app
(cd tcMenuGenerator/ && mvn -DskipTests clean install -f pom.xml)

# Run
(cd tcMenuGenerator/target/jfx/app/ && GDK_BACKEND=x11 java --module-path ../deps --add-modules com.thecoderscorner.tcmenu.menuEditorUI com.thecoderscorner.menu.editorui.MenuEditorApp)
```

I'll investigate how the app can be packaged.

moritz89, 2020-04-26:
Packaging currently fails under Ubuntu 20.04 due to [invalid hashes of java.xml](https://bugs.launchpad.net/ubuntu/+source/openjdk-14/+bug/1868699). I will follow up once that bug has been resolved.

davetcc, 2020-04-27:
Let's leave this open until we get to a point where at least one means of simple pacakging is working and can be repeated going forwards. 

Desired outcome: At least one method of packaging the Java application for Linux that works on common distributions. Additional instructions for other flavours that describe how to package.

ptapping, 2020-04-27:
I spent a bit longer looking at getting the plugins to load once packaged into a .jar (using maven assembly plugin), but gave up. The problem is I can't seem to get the Configuration/ModuleFinder to resolve the plugin packages and/or dependencies from inside the .jar, even though the documentation says ModuleFinder should work with a .jar passed in as a path.

So plan B is a bash script to do effectively what the java packager would do. Copy files to a (system or user) directory, then make a script in the path to run the java application. Pull request at #78

davetcc, 2020-04-30:
This is great news, along with this I've given some thought to the long term sustainability of the Java version. Given Windows 7 and 8 users will be around for at least another couple more years, and the Linux version works pretty much out of the box, I've taken the call to standardize the plugin system across the new C# and Java version. The C# one is MUCH better. You'll see there's a new dev branch with some heavy lifting going on there. ETA about a week to finish it, as it's easier to do than I first thought.

Once this is done it will use the XML plugins in the CoreXmlPlugins folder. These can reload at runtime and are served from thecoderscorner.com, meaning they will keep themselves up to date without needing constant re-releases.

So once done they should be very close on features:
Windows Store -> UWP version for Win 10
MacOS Store -> MacOS cocoa native version
Windows 7 & 8 and anyone with a Store dislike -> Java version
Linux -> Java version

This will make it very easy to keep both around for at least a few years. It will also mean it's easier to support mbed, particle, and other requested frameworks on both without any users missing out on much.

davetcc, 2020-05-02:
So there's a significant change on master, but it actually should make it much easier going forward to build the Java verison. The Java version is no longer dependant on pre-built plugins and the embedded directories. Instead, the plugins are now updated at runtime from thecoderscorner.com (this is copied from the store versions - they are no longer JAR executables, instead they are now XML and associated Arduino source code), but if you really don't want to do that, then you set command line VM option -DadditionalPluginsDir=<someDirYouLike> and put the plugins there instead. The UI will still tell you when there's a new version.

Another change is JDK moved from Java 11 -> Java 14 in the maven build file. My reasoning here is that Java 14 has a proper packager built into it so there should be fewer bugs with it (as it's not a backport). It should make little difference to the script (other than replacing the Java version from 11 to 14).

I'll have a look at the scripts tomorrow to remove reference to building the plugins, change the Java version to 14 and indicate how to set the system property if needed.

davetcc, 2020-05-12:
@ptapping and @moritz89, please could I trouble you once again, by asking you to try the new version with the adjusted instructions in the pull request below. If it doesn't work, please put in comments and I'll have to look at creating an Ubuntu Linux VM on one of my machines.

https://github.com/davetcc/tcMenu/pull/81

Going forward these changes make the Java version viable for at least another couple of years on Linux, Windows 7 and 8, because it uses the same plugin system as the Windows and Mac store versions, so there is far less maintenance overhead. I'm sure after you've tried it, you'll prefer the new plugin system too. Easier to edit, easier to build and easier for us to deploy

moritz89, 2020-05-12:
Sure, I'll test out the changes. And the move to Java 14 makes sense from a packaging perspective

davetcc, 2020-05-12:
If there were to be problems obtaining java 14 on any distro, you could still use Java 11, I actually realized that to package it using 14 can be done remaining compatible with Java 11. So I put the POM's back to 11 which means that you can choose either 11 or 14. Personally, I'd go with 14 if you can.

davetcc, 2020-05-12:
Many thanks by the way too.

ptapping, 2020-05-12:
OK, I grabbed that davetcc-linux-changes branch, modified the pom with `<version>1.4.0</version>` so that tcMenuJavaAPI and tcMenuCodePluginApi can get downloaded from the maven repositories, and also changed the javafx version and maven compiler versions to 11 to match my current default java version. Then ran the `install_linux.sh` script, and it compiled, install and ran  fine. Got the message to update arduino libraries, and clicked the button to download the plugins. Generating menu code seemed like it went fine, though I don't have my arduino with display here right now to actually test it.
I'll try with java 14 too just to be sure, but so far it looks good!

ptapping, 2020-05-12:
OK, well there's only a java 13 in the Arch main repositories at the moment, but it built fine with that too (with same changes to pom above to specify java 11). So maybe change the run-on-linux.md to specify "Java OpenJDK 11 or greater", and change the lines in the instructions to be `jdk11-openjdk` and `java-11-openjdk`.

The other issue I see with the instructions for checking out master is that then inside the pom it specifies an `x.y.z-SNAPSHOT` version requirement, which isn't available for maven to download, so the build fails. That's why in the older instruction I specifically switched to a commit/tag where the version was available on the maven repositories. This could be addressed with an extra `sed -i -e 's/\([0-9]\)-SNAPSHOT/0/' pom.xml` line, but that will downgrade to the x.y.0 release. That might work most of the time, but isn't perfect, and some shell scripting magic to replace with z-1 may still not work if version x.y.z-1 is not on maven repositories either. So maybe manually editing with "latest version on https://mvnrepository.com/artifact/com.thecoderscorner.tcmenu/tcMenuCodePluginApi" might be needed.

davetcc, 2020-05-13:
Thanks for checking this, I'll make the changes you suggest and merge the PR back in. I think the guide should only cover using the latest release build. I'll remove any references to other branches.

tcMenuCodePluginApi: this was completely removed from the codebase in 1.4.1 release, if there's any leftover references to it, I'll remove from the maven files. Building the designer should now just be "mvn clean install" on the latest release version of the tcMenuGenerator. Let me double check everything then update the docs

davetcc, 2020-05-15:
I had a go at installing virtual box and then trying to load Ubuntu LTS into it. My first attempt at this was so unstable that I think I just need to delete the virtual-disk and start again. I'm really familiar with Linux, but only server-side in headless mode.

When I get a chance I'll do a bit more reading on display drivers etc before trying again. My aim is to try and package it for at least Ubuntu/Debian.

davetcc, 2020-07-18:
I'm going to close this now, I think that the Linux instructions are clear enough for the moment, and it's got a whole load easier since the plugin download was automated into the tool. In future releases I'll add a zip file with all the jars to the release distribution, so Linux users can skip the build step.

closed 2020-07-18 by davetcc

moritz89, 2020-07-18:
Really cool. I have to say, that I’ve moved back to using neu-rah’s ArduinoMenu since they’ve changed licenses to LGPL. However I’m sure all other Linux users will appreciate the support 👍🏼

davetcc, 2020-07-18:
Whatever is good for you. We’ve got some very large changes coming up.
Maybe we’ll welcome you back at some point. Don’t forget tcMenu will always
be Apache licensed and commercial friendly.

combs, 2020-12-12:
hello! I'm running Ubuntu 20.04 and gave the most recent release a try. (caveat being that I know nothing about java, apologies.)

the startDesigner.sh makes the assumption that it is being run from the repo root; if you run `./startDesigner.sh` from inside `app` then the `pushd` yacks out. Suggest something like `cd "$(dirname $0)"` to handle path changes. and not sure you need to pushd/popd inside this script, since the shell script's environment's working directory doesn't take over the parent process's (as a .bat would).

It initially failed to start with this:

```
===================================================
Starting TcMenu Designer
Ensure that you have at least openJDK 11 installed.
===================================================
~/Dropbox/Git/tcMenuDesigner-1.7.0/app ~/Dropbox/Git/tcMenuDesigner-1.7.0
Graphics Device initialization failed for :  es2, sw
Error initializing QuantumRenderer: no suitable pipeline found
java.lang.RuntimeException: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer.getInstance(QuantumRenderer.java:280)
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.init(QuantumToolkit.java:244)
	at javafx.graphics/com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:260)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:658)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:409)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:1051)
Caused by: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:94)
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:834)
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:1051)
Caused by: java.lang.RuntimeException: No toolkit found
	at javafx.graphics/com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:272)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:658)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:409)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
	... 5 more

```

I got it a little further along with installing Ubuntu package `openjfx` but now it's yacking out in a way I can't puzzle out:

```
java.lang.ClassNotFoundException: com.sun.glass.ui.gtk.GtkPlatformFactory
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at javafx.graphics/com.sun.glass.ui.PlatformFactory.getPlatformFactory(PlatformFactory.java:42)
	at javafx.graphics/com.sun.glass.ui.Application.run(Application.java:144)
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.startup(QuantumToolkit.java:280)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:269)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:658)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:409)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[... cut ...]
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
	... 5 more

```

This is with release 1.7.0. I am running openJDK 11 on a Dell Latitude e7470 with Intel built-in graphics, in case there's any GPU requirement. 

My apologies if there are launch changes in the pipeline that haven't made it to a release yet; I wasn't sure how to interpret the above comments in this issue. 

Happy to open another issue if that'd be best. Many thanks, I am excited about this library and appreciate you releasing it.

davetcc, 2020-12-13:
Please feel free to raise an issue for the incorrect path and script problems.

However, it's not possible for us to help with core JDK issues, I suspect you've not fully installed a graphics capable JDK (or there are incompatible libraries in the JDK you have installed). There are two ways you can install a JDK headless, and graphics cable. I suspect your installation is either headless or broken looking at output provided above. We provide a Linux build on a best efforts basis that would work on top of a JDK that had graphical support.

Unfortunately, I have to say, even as a long-term user of Java on the server-side, that these days, it's getting harder and harder to get a working graphics cable JDK on Linux, and on Windows and macOS we have moved to dotnet Xamarin based UIs.

I would recommend you start with the simplest JavaFX or Swing app possible and see if you can get it to run on your JDK. 

Also notice in those errors you posted that there is not a single class starting with "com.thecoderscorner", it's failing at startup before it loads the main class, trying to initialise the graphics libraries. We do use Linux quite a bit server side, but have no experience trying to get graphics working as our Linux boxes are all servers.

davetcc, 2020-12-14:
I am assuming that you loaded the java / javafx pacakges from your distro's package manager, I would recommend bringing this up on the forum associated with the distribution. It looks like their distributions may be broken - I suspect, but cannot prove that the version of GTK library expected by the JDK is different from what you've got installed.  Maybe some support packages are needed that are not automatically installed.

davetcc, 2020-12-14:
As I said in the last mail, it seems this is caused by a missing dependency, I found this issue where it appears to have been fixed in the debian package, but maybe if you somehow get the package they reference installed, the JDK will work on ubuntu too.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=910585

combs, 2020-12-15:
Thank you--I really appreciate your gracious help with this. I now understand that any errors outside the "com.thecoderscorner" class are probably something wrong on my end. I'll keep chewing on my installation. (It's a desktop build of Ubuntu 20.04, after a few upgrades, if you were curious.)

davetcc, 2021-01-25:
Sorry, it's taken a long time to get around to this we are caught up in a major library update. I think I finally have the answers now. Please see the following links:

* https://www.thecoderscorner.com/jforum/posts/list/0/92.page
* New open issue that will fix this so the github clone is not needed https://www.thecoderscorner.com/jforum/posts/list/0/92.page