TcMenu/tcMenu #77: The conversion process has failed with an error: Header Generation failed

0sander, 2020-04-26:
Hi Dave,

I am having an issue generating code on MacOS with TcMenuDesigner 1.3.5. When generating code I get the following error message:

21:50:23 - Starting Arduino generate: /Users/xxx/Documents/Arduino
21:50:23 - Writing out header file: /Users/xxx/Documents/Arduino/Arduino_menu.h
21:50:23 - Failed to generate header file: Comparison method violates its general contract!
21:50:23 - ERROR during conversion---------------------------------------------
21:50:23 - The conversion process has failed with an error: Header Generation failed
21:50:23 - A more complete error can be found in the log file in <Home>/.tcMenu
 
I am attaching the emf file causing this error.
[decoder.zip](https://github.com/davetcc/tcMenu/files/4536352/decoder.zip)

Here is the log:

`2020-04-26 21:37:30 INFORMATION com.thecoderscorner.menu.editorui.project.FileBasedProjectPersistor open Open file /Users/ove/Documents/Arduino/decoder_final_mit_led.emf
2020-04-26 21:37:30 INFORMATION com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 0
2020-04-26 21:37:33 INFORMATION com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl showCodeGeneratorDialog Start - show code generator dialog
2020-04-26 21:37:34 INFORMATION com.thecoderscorner.menu.editorui.generator.ui.DefaultCodeGeneratorRunner startCodeGeneration Starting conversion for [Arduino AVR/Uno/Mega] in path [/Users/xxx/Documents/Arduino]
2020-04-26 21:37:34 INFORMATION com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator logLine Starting Arduino generate: /Users/xxx/Documents/Arduino
2020-04-26 21:37:35 INFORMATION com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator logLine Writing out header file: /Users/xxx/Documents/Arduino/Arduino_menu.h
2020-04-26 21:37:35 INFORMATION com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator logLine Failed to generate header file: Comparison method violates its general contract!
2020-04-26 21:37:35 INFORMATION com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator logLine ERROR during conversion---------------------------------------------
2020-04-26 21:37:35 INFORMATION com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator logLine The conversion process has failed with an error: Header Generation failed
2020-04-26 21:37:35 INFORMATION com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator logLine A more complete error can be found in the log file in <Home>/.tcMenu
2020-04-26 21:37:35 SCHWERWIEGEND com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator startConversion Exception caught while converting code: 
com.thecoderscorner.menu.pluginapi.TcMenuConversionException: Header Generation failed
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator.generateHeaders(ArduinoGenerator.java:327)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator.startConversion(ArduinoGenerator.java:117)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.generator.ui.DefaultCodeGeneratorRunner.lambda$startCodeGeneration$0(DefaultCodeGeneratorRunner.java:52)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.IllegalArgumentException: Comparison method violates its general contract!
	at java.base/java.util.TimSort.mergeHi(TimSort.java:903)
	at java.base/java.util.TimSort.mergeAt(TimSort.java:520)
	at java.base/java.util.TimSort.mergeCollapse(TimSort.java:448)
	at java.base/java.util.TimSort.sort(TimSort.java:245)
	at java.base/java.util.Arrays.sort(Arrays.java:1515)
	at java.base/java.util.ArrayList.sort(ArrayList.java:1749)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.generator.arduino.ArduinoGenerator.generateHeaders(ArduinoGenerator.java:308)
	... 3 more
`
What could be causing this problem and is there a way to fix it in the emf file?

Thanks!
ove

0sander, 2020-04-26:
The code generation seems to exit after while creating the header. The last lines written to the header file are:

// Callback functions must always include CALLBACK_FUNCTION after the return type
#define CALLBACK_FUNCTION

Callback functions are missing.

davetcc, 2020-04-27:
Looking into this failure now and can recreate locally. I'll feedback with a workaround shortly.

davetcc, 2020-04-27:
As an aside if you're writing to ROM directly in a callback - looking at SaveToRom() here. It is very, very dangerous. If any of those items are edited by a rotary encoder **it will break the EEPROM in days.**

I'd recommend setting a save to rom flag and writing it out only when there's a change, quite infrequently. IE in a task or something like that. The absolute best way to handle it is to have a Power loss detection circuit such as described here: https://www.thecoderscorner.com/electronics/microcontrollers/psu-control/detecting-power-loss-in-powersupply/

davetcc, 2020-04-27:
We found the bug and it will need to a code fix. Fortunately, 1.4.0 was going to be released very soon (today or tomorrow hopefully), this is a small low-risk fix so will patch it in. This also works on the new Windows Store version of the UI, if you are on windows you could use that now: For that see https://github.com/davetcc/tcMenuLib section about UI installation.

0sander, 2020-04-27:
Thank you very much Dave for looking into it and fixing this so incredibly fast!

I am aware of the fact that frequently writing to EEPROM will destroy it and came across your power loss detection circuit unfortunately too late, as my current prototype PCB does not have this circuit. My current solution is what you suggested. I am using the callback only to set a flag and then save once in while.

Would it be possible to have another callback that is called only when the menuitem has been left after a change? So that it is called not everytime a change is made but only when the change is committed by the user?

davetcc, 2020-04-27:
> Would it be possible to have another callback that is called only when the menuitem has been left after a change? So that it is called not everytime a change is made but only when the change is committed by the user?

I think what would be doable quite quickly is to add an OnItemCommitted(int id) to the menu manager. Would that work, if so it could be added in a subsequent 1.4 patch release?

Could you raise an issue in https://github.com/davetcc/tcMenuLib for this and I'll try and get it into one of the patch releases?

0sander, 2020-04-29:
Yes, that would be perfect. I am adding an issue to tcMenuLib.

Thanks for adding this!

davetcc, 2020-05-02:
Sorry, the release process for the Java version has taken a little longer than we expected this time around, a critical task slightly overran. We will be releasing very soon (Monday hopefully) pending all tests passing during the test cycle.

davetcc, 2020-05-04:
This should be fixed by either the windows store version or the latest Java version. Please reopen if otherwise.

closed 2020-05-04 by davetcc