TcMenu/tcMenu #88: MenuItem.getID() returns a different ID to the one shown in the designer

nickb512, 2020-07-03:
It could well be a PICNIC kind of issue, but if I run the following line of code I get back a totally different ID to the one shown in the designer. Is there another way to get the ID from the designer? The ID that is returned seems to change if I add a menu in.

`uint16_t currentMenu = menuMgr.getCurrentMenu()->getId();
`
As an example when the root menu is displayed (ID 0 in the designer) returns 7.

davetcc, 2020-07-03:
Ah yes that is something that doesn’t seem to make sense at first, but once you understand the structure on the device it suddenly makes sense. 

I’m soon going to improve the docs to better explain this. 

On the device the root menu item doesn’t exist, the first menu item is the first item under root in the designer, and then all the other root items are in a linked list under that. Each sub menu has a child member and that just works exactly the same, IE a linked list of menu items. The current menu root always points at the first item in a linked list. 

If you take a look at MenuItem in the reference docs you’ll see it has a getNext method and the SubMenuItem has a getChild

I hope this makes sense.

davetcc, 2020-07-03:
TASK: improve the documentation on theCodersCorner website to fully describe how menus are stored both on-device and in the Java API.

nickb512, 2020-07-04:
Thanks for confirming its not a bug. I have used the SubMenuItem and I can get the ID of the first item within the sub menu but what I am after is the ID of the menu we are currently in, not the one we have selected etc. In my Root menu I have a SubMenuItem called Status with an ID of 12. On the device when I scroll down to the Status menu and press OK to select it at this point I want to get 12 returned. 

I have tried a few different methods and can't seem to find it but its probably just me being thick.

davetcc, 2020-07-04:
Take a look at MenuIterator.h in either the reference docs or the source code, it has methods to get the parent amongst others. I’ll reply again later with more detail as I’m replying from my phone.

davetcc, 2020-07-18:
TASK: Provide a helper function that gets the submenu that is currently being rendered rather than the first item, probably on menuMgr. However, an open question is what should it do when root is being rendered as it doesn't have a parent?

davetcc, 2021-01-03:
This will be available in tcMenuLib V1.8 due shortly as `MenuItem* getSubMenuFor(MenuItem* current);` in MenuIterator.h

closed 2021-01-03 by davetcc