TcMenu/tcMenu #89: Change or Hide Title in runtime code

nickb512, 2020-07-03:
In the project I am currently working on we don't need a title row displayed at the top, but the first menu item is a 'state' read only text field. It would be great to either be able to hide the title row or change the title text throughout the sketch to display this state information

davetcc, 2020-07-03:
Depending on which display you are using, some of them have a configurable title bar. Basically all displays based on liquidCrystal have this support as does the ssd1306ascii display. For these two displays you will see methods to turn off the title bar. Which display driver are you using?

nickb512, 2020-07-04:
Excellent news, thank you. I had looked all through the LiquidCrystalIO library but forgot to look in the tcMenuLiquidCrystal files inside the project. Found it now and the pesky title is gone. Thank you for a great menu :)

closed 2020-07-04 by nickb512