TcMenu/tcMenu #92: “tcMenuDesigner.app” is damaged and can’t be opened.

DeeEmm, 2020-08-05:
Trying to run the latest TC Menu designer 1.5 but when I open the file I get the following message

“tcMenuDesigner.app” is damaged and can’t be opened. You should move it to the Bin.

I have tried re-downloading the file and reinstalling but get the same result.

I'm running OSX Catalina 10.15.5 (19F101)

/DM

davetcc, 2020-08-05:
This is because Catalina requires all apps to be Notarized. Please see the linked issue below:

Please see: https://github.com/davetcc/tcMenuLib/issues/65

We are actively testing a new version of the generator that will be either Notarized or available on the App Store. It's going through final beta today and if that completes without more bugs, we may just push it with a few very minor known issues, as we've got quite a few requests to get it out now.

DeeEmm, 2020-08-05:
1.3.5 is working fine. It's just later versions: 1.4.2 / 1.5.0

DeeEmm, 2020-08-06:
Haha, looks like I originally posted the fix linked to in that post lol. Been a while since I've needed to use the app. Memory is not as good as it once was.

Had a user that flagged an issue which made me take a look at it. (more on that in another post).

DeeEmm, 2020-08-06:
Thanks Dave. 👍

davetcc, 2020-08-18:
Application is now available in the Mac App Store, "tcMenu Designer":

https://apps.apple.com/gb/app/tcmenu-designer/id1527782002?mt=12

I'd be glad to get any feedback on it. It's a native Cocoa app that's based on the same generator as the Windows store generator; which as been around for a while.

davetcc, 2020-08-18:
I do most of my C++ / embedded development on my Mac, so the designer has been quite well tested!! We've also put it through a full beta cycle as well, where we even found a few bugs in the existing windows one.

DeeEmm, 2020-08-18:
Downloading now. Will give it a test drive and let you know how I get on.

DeeEmm, 2020-08-18:
Works great Dave, thanks for all of your hard work.

davetcc, 2020-10-26:
Unfortunately, the App Store version is getting some poor reviews and we don't have enough data to know if it's working properly, linked with #104 as we may need to temporarily remove the App Store version unless we get some real world data..

DeeEmm, 2020-10-26:
I just added a review for you. 👍 

All working okay here. What kind of issues have been reported? I can try and replicate them on my setup if you need some tests done. I'm limited on display types however.

DeeEmm, 2020-10-26:
BTW, if you remove the app then my end users cannot use my project.

davetcc, 2020-10-26:
No we would not fully remove, we just discussed going back to the old java one while we notarised it in the worst case. The issues seem to be around problems with file permissions when the generation directory is not sketches on the store

davetcc, 2020-10-26:
Don't worry we are committed to a Mac version, being a Mac user myself. We just may need to re-awaken the java version for a week or so while we move to notarisation, if we cannot get the store version bullet-proof in these cases.

davetcc, 2020-10-26:
In V1.7 coming out later this week we've made it harder to misconfigure and more resilient to the user accidentally denying file permissions when generating outside the regular sketches directory. But we don't see any instability here of the type the two users with the 1* reviews indicated, one of the 1* reviews has since been removed.

DeeEmm, 2020-10-26:
Yeah it kind sucks when people downvote stuff but give no feedback. Not very helpful that. 

I've left some comments in #104 that may / may not help. :)

davetcc, 2020-10-26:
I've raised a developer support ticket with Apple too, asking them how Xcode handles writing these files into the project directory without need a save dialog.

DeeEmm, 2020-10-26:
Yeah Catalina has really changed how things work. I'm doing a lot of python development at the moment as it's constantly asking for file access permissions.It's a PITA. 

From what I can see, the app only requires approval for access once and then it is remembered.

DeeEmm, 2020-10-26:
(BTW I'm not specifically talking about TCMenu there)

davetcc, 2020-10-27:
Closing out this issue. I've already done a test and it looks so trivial to notarise the current App Store app that this option should not  be needed in the worst case. With V1.7 I showed it to my tester and he was surprised how much easier it was to setup. We also found the bug that I think they hit with spaces in directory names, that was not related to usability.

closed 2020-10-27 by davetcc

DeeEmm, 2020-10-27:
I'll give it a test drive once it's published.