TcMenu/tcMenu #93: GFX Renderer Issue with Custom Fonts

betti154, 2020-08-10:
Using custom fonts priorit to entering the menu, screws up the GFX renderer after exiting menu. As per:

https://www.thecoderscorner.com/jforum/posts/list/0/63.page?OWASP_CSRFTOKEN=5O4U-84AR-VJMM-12CW-DERN-DUSI-R00L-O4Q4#p317

A simple call to gfx.setTextSize(1) seems to reset things enough to then set custom fonts again and have things work normally. 

Workaround solution easily managed, I'm raising this bug as requested by Dave. 

Thanks, Damien

davetcc, 2020-09-15:
I've tried this now with a modification to the example colorTftEthernet32. It takes the display, re-initialises the font sizes, and does some rendering. On holding the button it gives the display back. It seems good in all cases.

davetcc, 2020-09-16:
Please see the example in the previous comment, colorTftEthernet32 I added a test case to there and it appears to work as expected. If you think there's still a broken case, could you tell me what's missed in that example here.

davetcc, 2020-09-22:
I've finally managed to recreate this (at least I think). It's in the way that set cursor seems to work when the text size or custom font is used.  I think re-ordering the font, text size and cursor operations will sort this out. IE: 

1. reset the font
2. reset the font size
3. set the cursor

davetcc, 2020-09-24:
We've moved the plugins to their own project. It's too hard to release them otherwise. 

Moved to https://github.com/davetcc/tcMenuXmlPlugins/issues/2

closed 2020-09-24 by davetcc