TcMenu/tcMenu #94: FloatAnalogItem - wholeAndFraction is unable to set neg. values between 0 and -1

watzi3, 2020-08-17:
Using an FloatAnalogItem and decrementing it under 0 (in increments smaller than 1) sets the value still positive. 

Function found in: 
MenuItems.cpp at line 149

```
float AnalogMenuItem::getAsFloatingPointValue() {
	WholeAndFraction wf = getWholeAndFraction();
	float fract = (float(wf.fraction) / float(getActualDecimalDivisor()));
	return  (wf.whole < 0) ? (wf.whole - fract) : (wf.whole + fract);
}
```
example: 
Current value = 0 --> decrementing it by 0.5 returns 0,5 and not -0,5. 
Problem: The Whole-value can not become -0 and the following return conditon returns the calculation of the "false" condition: 

`return  (wf.whole < 0) ? (wf.whole - fract) : (wf.whole + fract);` 

Additional error: 
Even if it is possible to return a negative value between 0 and -1, it still cannot get displayed negative on the lcd.

davetcc, 2020-08-25:
MOving the issue to the tcMenuLib repo. See https://github.com/davetcc/tcMenuLib/issues/71

davetcc, 2020-08-25:
There's no choice here than to do a breaking change, it was an oversight during original development. I will add a negative flag to the whole and fraction. I'll keep breakages to a minimum.

Unit tests added and code being fixed. Will make a patch soon.

davetcc, 2020-09-04:
Fixed in latest tcMenuLib, although the change is unavoidably slightly breaking, it should be nearly transparent to >90% of cases.

closed 2020-09-04 by davetcc