TcMenu/tcMenu #95: Compile error on PlatformIO / STM32F103C8

Stanelec, 2020-09-10:
When compiling the generated code in PlatformIO, it occurs the same error as described in https://github.com/davetcc/tcMenuLib/issues/72.

The error is caused by this line of code:
https://github.com/davetcc/tcMenu/blob/739f68ef9cfa774d5bd03deb007e33515f3782ca/CoreXmlPlugins/core-display/lcdSrc/tcMenuLiquidCrystal.cpp#L85

I think it should be:
```c++
 uint8_t last = min<int>(bufSz, strlen(buffer)); 
```

Thank you for your work!

davetcc, 2020-09-10:
We’ll get the two compile errors in a patch release soon. Thanks for reporting.

davetcc, 2020-09-15:
please could you go to the designer, where the library versions are shown, change the stream from stable to beta, and check if you still get the compile error in liquidcrystal code. We've also update tcMenuLib master with a fix too.

davetcc, 2020-09-16:
I'm pretty sure this is fixed in plugin version 1.5.1, please try that version (beta stream at the moment) and re-open if I am wrong.

closed 2020-09-16 by davetcc

Stanelec, 2020-09-18:
You're right, everything is compiling fine now, thank you!