TcMenu/tcMenu #104: Need urgent help from MacOS users for testing designer

davetcc, 2020-10-26:
I’ve just been in a meeting with the board and they are really not happy with the state of play on the MacOS App Store app. I’ll be honest, we just cannot leave it in the store the way it is.  We really need to get some quite urgent feedback. There appear to be several very negative experiences without any feedback here or on the forum. 

In Windows the sand boxing was far easier to work with and the app fits well in the store. However, I’m not sure if it is a good fit for the App Store, it seems harder to fit it within the security scopes. We are even open to if people would prefer a notarised app. 

If you use tcMenu on MacOS I ask that you try the App Store version and report back any issues you find either here or in the forum. We ask for your help because we don’t see this instability in our testing.

In the next version we have doubled down on initial setup. In 1.7 it includes enough plugins to get started and won’t let you get to the generator until the projects directory is configured. It also does more to make sure that writing outside the sandbox is managed.

davetcc, 2020-10-26:
We suspect from the very limited details left by the user in the negative review (another negative rating has no review) that they ran into difficulties because the code generator was run outside of the project folder (or the project folder was not set). We tried it ourselves and although it doesn't crash, it is possible to end up with no code generation.

Maybe the nature of this app doesn't lend itself to the Mac App Store. I'm going to open a developer support ticket with Apple, with this we can see if there's anything we can do to improve this edge case by better use of their APIs / permissions.

In the meantime, any real-world feedback would be greatly appreciated, no matter how negative.

DeeEmm, 2020-10-26:
From the outside looking in, if it's possible to misuse the app, then it's not an edge case it's a usability problem. Generally in software, if something can happen, it will. So all possible failure modes should be considered and addressed.

Either the app needs to detect such cases and report it to the user (app in incorrect location), or the app needs to automagically create the correct folder structure for the user, or failing that the app needs to walk the user through the process step by step. 

If you are relying on the end user to read the documentation, then you can guarantee they will not.

The app works for me, and it works for you because we know how to use it, so it is almost impossible for us to use it incorrectly as we know that it will not work.

However, there is a bit of a learning curve with using the app - https://www.thecoderscorner.com/jforum/posts/list/30.page

That's my initial foray into using it, and that's after reading the documentation. I've also been a software engineer for most of my life, so I can say that it is definitely not super intuitive. (sorry, but that's just how it is.) 

Anyway, not trying to bash, just hoping to help.

/DM

davetcc, 2020-10-26:
Agreed, 1.7 takes a different approach at startup. It won't allow code generation until it detects that at least the sketches directory has been set, it also packages the plugins with the app, so there's relatively recent ones there when it starts up. Along with the stricter checking I hope it will at least prevent the full failures that they saw. A bit like any IDE, there's a level of complexity that just cannot be hidden away, and the strict sandboxing situation with the App Store adds another layer of complexity. I'll wait for the reply from Apple and testing after 1.7, but worst case we'll just move the same app to be notarised. It should be very simple as it's just AppStore-- basically

DeeEmm, 2020-10-26:
Sing out if / when you need testing.

davetcc, 2020-10-27:
So 1.7 should come along soon, let's hope this can right some of the wrongs...

1. refuses to generate when sketches directory not set - with helpful dialog.
2. At start up it provides a kind of "todo list" of things that need to be done to make it work, with a go to preferences button underneath. 
3. It now packages plugins as an embedded resource, so starts with relatively recent ones automatically. Still allows for updates within the app, but they are there immediately on start up at least.
4. tcMenu library is in both platformIO and Arduino library manager these days, so it's installed there, dependencies are automatic. The todo list tells users to use library manager to install.
5. If the project directory has spaces in it, code generator won't start and indicates that toolchains don't like spaces. (Arduino really doesn't).
6. If the sketch directory is outside of the sketches directory, then we've doubled down on the sandbox elevation code, to prevent it ever trying to run without permission being obtained, and provide better errors about what went wrong.
7. We've made many of the dialogs a bit clearer.

Also, for a while now we've had code that handles plugin files being added and removed (EG change of display), to avoid conflicts on that front, so that shouldn't be an issue. The only other think I can think of is if the directory name is changed, it may cause some issues. I'll maybe try and get that into another patch release. 

If you can think of anything else that would be fairly quick, let me know..

davetcc, 2020-10-27:
> If the project directory has spaces in it

This was it I think, it acted exactly as per the review. tcMenu was using NSUrl.FromString() instead of NSUrl.CreatePath. I've fixed this, but the thing is that using a directory with spaces in it, that will create filenames with spaces in them is bad news. But it doesn't crash now, it produces a warning.

DeeEmm, 2020-10-27:
Is the directory path from user input or code generated?

For user input with NSUrl.FromString() you would need to sanitise the input and ensure that other non-valid characters are also excluded, which is not to say that you should not be doing this for NSUrl.CreatePath as well. 

At a GUI level, input fields should have input validation performed, which may need to be done programmatically i.e. a textbox is not a 'URLbox' (IIRC it will not automatically validate a URL type input). It is probably still possible to input non-valid characters like ^*( etc. So they need to be tested for and results reported back to the GUI if the test fails.

This is a first level of defence, but usually not infallible.

Then you need to sanitise the input that is received from the form. The best method is to use a regex string replace and remove all non-standard characters, apart from space, which you replace with underscores.

With web apps / web pages this is standard practice to prevent attacks from SQL injection, but with an installed app it prevents crashing due to user stupidity. :D

davetcc, 2020-10-27:
Agreed, if it was not in the App Store I would literally lock down where and what could be selected in the save dialog with any advanced options hidden by default. Unfortunately, while it's on the app store I cannot take control of the save as dialog which is provided by the OS, so we have to let them save it where they want to - on both Windows and MacOS store apps, permissions can only ever be gained through the open and save dialogs. The only thing I can do is validate later that it's likely to work with Arduino, before code generation; which is what I've added for space which is the most common mistake.

I'm going to look into what filtering is possible within the save dialog. To see if there's some way I can provide a filter that excludes anything except alphanumerics and underscores. But I'll do this after 1.7 gets out. I think there's enough in 1.7 that it is worth getting out.

Another possibility is to have two save options, a safer "save into sketches" that just takes a carefully filtered name and uses the existing permissions already obtained for the sketches folder. Not even sure if this would work though, I'd need to try and hook into the right events.

davetcc, 2020-10-27:
Thanks a lot for the suggestions BTW, it's really helpful at the moment to get as many ideas as possible.

DeeEmm, 2020-10-27:
No Dramas, I'm not sure how relevant my comments are, it's been an age since I've done any app development, I'm sure that the landscape has changed a lot.

As an aside, I think that OSX has major dramas with its save dialog at the moment. I get *LOTs* of issues with Chrome and other apps locking up when saving a file (I use Chrome a lot lol.) The only way out is to kill the app-save-dialog process via the activity monitor. This is on a New 27" iMac with a fresh vanilla OSX Catalina installation.

I've noticed that there are a lot of app-save-dialog artefacts present in activity monitor at any given time (basically from any program where it has been used).  To me it seems like the app-save process is not getting cleared down correctly by OSX after the file save process has been undertaken. The save action is completed and the save GUI closed, but the process is still active. Basically a memory leak issue.

If you have not considered this it may also be worth looking into.

/DM

davetcc, 2020-11-02:
We are on the edge of releasing TcMenu library and designer V1.7. From this point in the new focus is stability and usability. There are five things to do in order:

1. [DONE] A "Getting started with tcMenu" YouTube video for new users, https://youtu.be/RKywGOjiAjQ
2. [FINAL-TESTING] tcMenu library 1.7 to Library Manager / platformIO
3. [FINAL-TESTING] tcMenu Designer 1.7 for MacOS
4. [FINAL-TESTING] tcMenu Designer 1.7 for Windows
5. [Smoke test] Legacy tcMenu Designer for Windows 7 & 8. along with build of Java API

davetcc, 2020-11-04:
We are on the edge of releasing TcMenu library and designer V1.7. From this point in the new focus is stability and usability. There are five things to do in order:

[DONE] A "Getting started with tcMenu" YouTube video for new users, https://youtu.be/RKywGOjiAjQ
[DONE] tcMenu Designer 1.7.1 for Windows - after doing the release, we pulled two crash reports around deleting folders while the apps running, we've fixed those up.
[DONE] Java remote control API
[DONE] Legacy java UI for Windows 7 and Linux.
[Release being prepared] tcMenu Designer 1.7 for MacOS

Once above are complete: bump version of tcMenu library 1.7 to Library Manager / platformIO

davetcc, 2020-11-05:
Now submitted 1.7.2 to App Store. It contains a lot of additional error handling, can even deal with folders disappearing underneath it. Should be available in the next day. Legacy java UI released.

Once this is available, I'll bump the lib up to 1.7 too

davetcc, 2020-11-06:
1.7 is in the Mac store. All 1.7 versions are out. We we have tried to fix as many file system and new user usability issues as possible! Feedback most welcome.

closed 2021-01-03 by davetcc