TcMenu/tcMenu #105: tcMenuDesigner.app enum menu item  - doesn't  save item names

MarcusSRussell, 2020-10-29:
Very new to this app but loving it!

So I may have missed something but am observing the following:
Using version 1.5.10 on Mac Book Air running Catalina 10.15.7

There seems to be a small issue when you create an enum menu item.
You can add items and you can double click and edit. But if I add another item the text on all items resets to 'New Item' . Entering New Items and then editing their names only retains the names until you move onto another part of the menu tree - go back to the enum menu item and the item names are all reset to 'New Item'...

[image]
[image]
[image]

Obviously easy enough to work around by editing the generated files but would be nice to get sorted at some point

davetcc, 2020-10-29:
Hi there thanks for reporting it and the compliment too. We’ve had a few new users who had very negative experiences with the Mac designer UI recently so it’s good to hear positive feedback. 

 I’ve just taken a look over this and it was an easy fix to get 95% of the way there, leaving only pressing add item while editing as the one case to fix up later.

I’ll include it in 1.7 which is due very soon.

In the mean time you can press the enter key after editing and it will work.

MarcusSRussell, 2020-10-29:
Yep that works and is fine pro tem
The ability to build a menu system so quickly way outweighs any minor uI issues in my view
This library will save me a ton of time on my current project

MarcusSRussell, 2020-10-30:
I noticed earlier today your interest in things audio so thought worth mentioning my project is a passive balanced TVC preamplifier controlled by arduino with rotary and IR remote.
Given the ability of tcMenu to handle all the settings aspect (volume units, input names, input gain etc etc ) it greatly simplifies my approach which I thought would need to be state based but now looks too straightforward for that to be a benefit - I am just a hobby enthusiast although this will hopefully go into my main system which is CD, turntable and streamer based with tube amplification and electrostatic speakers

davetcc, 2020-11-06:
Could you try the 1.7 version please that has just been released. See if it works better now. 

There should be only one last case with pressing the add item button while editing. We’ll fix that shortly.

MarcusSRussell, 2020-11-06:
Will do Dave

davetcc, 2020-11-06:
BTW the writeup on your system is great, an integrated amplifier was the first use case for TcMenu, seeing what was available at the time led me to write it!

Let me know how it goes once you've got it fully integrated into your equipment.

MarcusSRussell, 2020-11-06:
Hi Dave

Please bear in mind with the following that I am no expert and entirely self taught so I may well use incorrect terminology etc.

I have installed tcMenu to another Mac (iMac) to give me a cleanish install.
Felt the splash screen was clear and I watched the video which was clearest although a little fats in places - but of course you can pause it.

The screen also prompted me to update the sketch folder and library paths. As the video expressly mentioned this it seemed very clear to me as to what to do and why.
As mine is a standard folder construction it correctly guessed library path.

It gave me a warning message about installing tcMenu and the supporting libraries within Arduino library manager.
Before I did that I checked the ;library versions which it showed as up to date. I assume the desktop program has its own copies of the tcMenu libraries? I did find this just slightly confusing although as I then updated my Arduino libraries to include tcMenu and its supporting libraries I guess anyone would get there anyway.

I then constructed an example library with no issues and the previous issue I had with not retaining enum entry item names is fixed. 
I did recreate the case where editing is lost if in the editing field and clicking on ‘add item’ although I think this is a pretty minor issue.

Lastly I notice the tree display looses focus when ever you enter or edit something within an item - this was present before and I have found that as my menu structure has grown larger for my preamp project this has annoyed me just slightly.

As mentioned by others and yourself I do wonder none the less wether this is an appropriate app for the App Store - as you do highlight in the store this is not for everybody and I do wonder what peoples expectations are of such an app unless they have a lot of knowledge in which case like me they would find it through seeking out a library solution rather than searching on the App Store. Plus with small volume a few negative reviews skew things in an unfair way.. 

Anyway keep up the excellent work.

Cheers
Marcus

MarcusSRussell, 2020-11-06:
I will

This is almost certainly not the right place to ask this question but I am planning to use an IR remote with this preamp and was originally planning to use it for menu navigation as well. I am using an encoder also and that is working fine with the menu and given how it all operates smoothly I will likely just use the remote for volume, input change, balance change and mute/off but I wondered if anyone has somehow implemented an IR remote with tcMenu?

Cheers
Marcus

davetcc, 2020-11-07:
> This is almost certainly not the right place to ask this question but I am planning to use an IR remote with this preamp and was originally planning to use it for menu navigation as well

I've started a new thread on the forum to discuss this in case you are interested: https://www.thecoderscorner.com/jforum/posts/list/0/77.page

MarcusSRussell, 2020-11-07:
Thanks Dave I’ll give those thoughts some consideration 
Marcus

closed 2020-11-26 by davetcc