TcMenu/tcMenu #106: font size not reset on  renderer.takeOverDisplay or CALLBACK_FUNCTION

MarcusSRussell, 2020-11-26:
I have been using tcMenu with OLED display and u8g2 library and all worked well.
I have now changed my project to using a TFT screen using Adafruit ILI9341 and GFX libraries.
When the  renderer.takeOverDisplay function is triggered it returns control and if a menu item which uses the CALLBACK_FUNCTION is used this in my case is also returning control after saving changed settings.
Both work fine but I am not using the standard font I am using alternate fonts and custom fonts.
After returning control those fonts are the wrong size (too large).
I have found that adding a setTextSize(1); in the function called by renderer.takeOverDisplay or CALLBACK_FUNCTION fixes this but I didn't have this issue with the u8g2 /SSD1306 set up so I assume this behaviour is not by design?

davetcc, 2020-11-26:
>I have found that adding a setTextSize(1); in the function called by renderer.takeOverDisplay or CALLBACK_FUNCTION fixes this but I didn't have this issue with the u8g2 /SSD1306 set up so I assume this behaviour is not by design?

Just trying to understand the problem, do you have an issue when you first take over display, IE when rendering your own display? If so, yes that is by design in Adafruit_GFX, before drawing any text, you need to ensure that you first call setTextSize with the appropriate size multiplier. Otherwise, it may well be wrong.

So if all your drawing uses the same font magnification, you would normally set the text size to 1 when you first take over the display.

I've checked over our rendering code, and as far as I can tell, it always sets the size before doing any text rendering.

MarcusSRussell, 2020-11-26:
Hi Dave
Thanks for the as ever prompt response.

I think this is because with Alternative fonts they are sized (8pt 12pt etc) as part of their include file - i.e. they are meant to be 8pt 12pt etc. but it appears setTextSize() overrides the standard size. The fonts I am using all have an 'innate' size so I am using them like that but I suspect maybe tcMenu is using a larger size on the TFT than setTextSize(1)? and when I take back control it is still using the last setting for setTextSize()?
As I say not a big deal anyway once I worked out how to get what I needed.

davetcc, 2020-11-26:
Yep, when you take control, you need to "reset" all the display settings such as font and size to the default. As we don't know what settings will work for you.

BTW: Even within the menu drawing, the sizes are configurable too, you can change the zooms to whatever you think works best, along with the color and padding around items:

take a look at: https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/rendering-with-tcmenu-lcd-tft-oled/#display-configuration-for-bitmapped-displays-adagfx-and-u8g2

MarcusSRussell, 2020-11-26:
That's excellent - I had failed to notice that and it will be very helpful!

davetcc, 2020-12-04:
Closed, functionality confirmed to work as expected, we don't know what settings to apply when takeOverDisplay is called, the font, size and colors should be reset before use.

closed 2020-12-04 by davetcc