TcMenu/tcMenu #107: part of tcMenu redisplays after renderer.takeOverDisplay

MarcusSRussell, 2020-11-26:
I am using a TFT screen using Adafruit ILI9341 and GFX libraries.
I have a couple of functions in use when taking back control.
First one is onMenuBeingReset and uses renderer.takeOverDisplay after setResetIntervalTimeSeconds of 5 seconds.
It returns control and draws my custom screen but after a fraction of a second it displays the list part of the menu over the current display but my main program still has control of the rotary encoder (i.e. the encoder does not have any impact on the tcMenu part of the overplayed display).

If I use a CALLBACK_FUNCTION to exit the menu (called... exitMenu) and take back control it all works as expected.

If I change the callback for renderer.takeOverDisplay from onMenuBeingReset to exitMenu I get the error again.

This suggests to me that something is not quite right with renderer.takeOverDisplay or I have missed something.

I did not have this error awhen previously using an OLED display with u8g2.

I have tried to capture this in the accompanying photos of the display.
[image]
[image]
[image]

davetcc, 2020-11-26:
I'm wondering if there is a race condition here, could you share the reset method here so that I can see what it's doing?

MarcusSRussell, 2020-11-26:
Hi Dave

Here it is:
_______________
void onMenuBeingReset() {
    
    Serial.println("Menu being reset");
    SetDisplayInputDetails(CurrentInput);
    gfxtft.setTextSize(1);
    CurrentState = VOLUME_CHANGE;
    CurrentScreen = MAIN_SCREEN_VOL;
    ScreenControl();

}
________________
ScreenControl determines what happens for each screen:

________________
void ScreenControl() {

  LastEncoderChange = millis(); //we are now going to a screen and if input or balance we will set a timer which is updated if the encoder changes
  DisplayTimeout = menuDisplayTimeout.getCurrentValue();//in minutes
  DisplayTimeoutMillis = DisplayTimeout * 60000;
  
  switch(CurrentScreen) //depending on screen we need to adjust the Encoder Precision
  {
    case SCREEN_OFF://no display State either IDLE or RUNNING_INVISIBLE
          //take control of screen and turn off
          Serial.println((String)"Screen OFF");
          renderer.takeOverDisplay(StateControl);
          drawMainScreen(NONE);
    break;
    case MAIN_SCREEN_VOL://State VOLUME_CHANGE,
          //Display volume screen set rotary encoder precision to volume max and current
          Serial.println((String)"MAIN_SCREEN_VOL");          
          renderer.takeOverDisplay(StateControl);
          drawMainScreen(UPDATE_VOL);
          if (menuVolumeUnits.getCurrentValue() == 0){//smoothed dB
            switches.changeEncoderPrecision(menuVolumeCurrentVolume.getMaximumValue()/2, (menuVolumeCurrentVolume.getCurrentValue()) / 2);
          }
          else if (menuVolumeUnits.getCurrentValue() == 1){//sets dB to range 0 - 22 to make it cleaner for use (almost full turn but less than the 4 turns otherwise although acceleration helps
            switches.changeEncoderPrecision(22, setdbVolfromCurrentVol(menuVolumeCurrentVolume.getCurrentValue()));
          }
          else{//%
            switches.changeEncoderPrecision(menuVolumeCurrentVolume.getMaximumValue(), menuVolumeCurrentVolume.getCurrentValue());
          }
    break;    
    case MAIN_SCREEN_INP://State INPUT_CHANGE
          //Display volume screen set rotary encoder precision to volume max and current
          Serial.println((String)"MAIN_SCREEN_INP");
          renderer.takeOverDisplay(StateControl);
          drawMainScreen(UPDATE_INPUT);
          switches.changeEncoderPrecision(menuInputsCurrentInputID.getMaximumValue(), menuInputsCurrentInputID.getCurrentValue());
    break;    
    case MAIN_SCREEN_BAL://State BALANCE_CHANGE
          //Display volume screen set rotary encoder precision to volume max and current
          Serial.println((String)"MAIN_SCREEN_BAL");
          renderer.takeOverDisplay(StateControl);
          drawMainScreen(UPDATE_BAL);
          switches.changeEncoderPrecision(menuVolumeCurrentBalanceDiff.getMaximumValue(),    menuVolumeCurrentBalanceDiff.getCurrentValue());
    break;    
    case SETTINGS_MENU_SCREEN://no display State either IDLE or RUNNING_INVISIBLE
         //Update state to Settings Menu and give back display
         Serial.println((String)"SETTINGS_MENU_SCREEN");
         renderer.giveBackDisplay(); 
    break;        
  }
}
______________

Assuming we are not giving back control it takes control and passes to renderer.takeOverDisplay(StateControl); which does all the work of changing and updating the main screen.
I hope that makes some kind of sense?

As I say I call another function which is really the same as onMenuBeingReset() but just updates variables for any changes within tcMenu system which have not been saved to eprom and that works fine.

davetcc, 2020-11-26:
Thanks, I'll need to try and see if I can recreate something similar.

MarcusSRussell, 2020-12-03:
@davetcc 

My sincere apologies for wasting your time on this - I found that I was drawing the screen before renderer.takeOverDisplay instead of drawing at the top of the function called by renderer.takeOverDisplay. Not sure why my CALLBACK function works ok but anyway correcting this has resolved this issue.

davetcc, 2020-12-03:
No problem. As it happened I’d been really busy trying to get the documentation and a couple of really critical issues finished off. So I had not got around to it.

davetcc, 2020-12-04:
Closed. Not an issue.

closed 2020-12-04 by davetcc