TcMenu/tcMenu #108: can't start on linux

diehard67, 2021-01-18:
when I try to run this I get this mess.
`diehard@bigred:~/Downloads/making stuff/arduino/tcMenuDesigner-1.7.0$ ./app/startDesigner.sh 
===================================================
Starting TcMenu Designer
Ensure that you have at least openJDK 11 installed.
===================================================
~/Downloads/making stuff/arduino/tcMenuDesigner-1.7.0/app ~/Downloads/making stuff/arduino/tcMenuDesigner-1.7.0
java.lang.ClassNotFoundException: com.sun.glass.ui.gtk.GtkPlatformFactory
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
        at java.base/java.lang.Class.forName0(Native Method)
        at java.base/java.lang.Class.forName(Class.java:315)
        at javafx.graphics/com.sun.glass.ui.PlatformFactory.getPlatformFactory(PlatformFactory.java:42)
        at javafx.graphics/com.sun.glass.ui.Application.run(Application.java:144)
        at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.startup(QuantumToolkit.java:280)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:269)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:658)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:409)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:1051)
Failed to load Glass factory class
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:1051)
Caused by: java.lang.NullPointerException
        at javafx.graphics/com.sun.glass.ui.Application.run(Application.java:144)
        at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.startup(QuantumToolkit.java:280)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:269)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:658)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:409)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
        ... 5 more
~/Downloads/making stuff/arduino/tcMenuDesigner-1.7.0
diehard@bigred:~/Downloads/making stuff/arduino/tcMenuDesigner-1.7.0$ 
`

any ideas as to how I can fix this??????
os is ubuntu 20.10

davetcc, 2021-01-18:
This is your Java installation. Unfortunately, running graphical Java applications on most Linux boxes has become very difficult.

The stack trace above is in JavaFX and there's not much we can do. As I understand it, there is a mismatch in libraries that leads to this, and unfortunately, it's beyond our control. Please take up with the distributor source.

Please notice that in the stack trace, there are no "com.thecoderscorner" packages, basically, the JDK failed to start.

davetcc, 2021-01-18:
I'll see if I can source an ubuntu install somewhere to try it out - may take some time though.

davetcc, 2021-01-21:
Update, our tester is going to set up an ubuntu virtual machine and run it up. We'll come up with a list of tasks that are needed to get it working. 

As an aside, and it really pains me to say it as a long term Java developer, to say the very least, JavaFX is causing us a lot of problems. So many that we moved new UI development to Xamarin; which was a massive and painful transition with bumps along the way. We'll look to move this last UI away from Java in the future, but we are very resource constrained and quite honestly struggling under the level of usage of the project.

For the moment we'll do our best to make it work.

davetcc, 2021-01-25:
There's a new entry on the forum showing how to run the designer on Linux. It seems to be working after following these instructions. https://www.thecoderscorner.com/jforum/posts/list/0/92.page

We will build the Linux distro on a Linux box starting from the next version, which is not that far away. In the meantime you can use the forum post.

davetcc, 2021-02-04:
Provide a tested Ubuntu distribution for Linux. Ensure the existing Linux instructions work for all other users.

davetcc, 2021-02-04:
We've just done a huge commit to ensure the Java UI is viable for quite a while to come. It brings it much closer to the same standard as the Windows and Mac store versions. Things we've added that will be in V2.0:

* Ability to easily override the Arduino directory from within the app (from Code menu)
* Ability to turn off global library checking when you have platformIO or similar build system (checkbox on main window)
* Menu item variable names are now locked down by default after the first time the code generator is run. You can choose to manually sync them back to the name at any time from the UI.
* More automated testing to ensure it stays working going forward.
* Catch up on Rgb and Scroll choice menu items.

davetcc, 2021-02-16:
There is now a proper version built as a Debian package, it will be available with every release, but is packaged in the Linux section of the below link while in beta:

https://www.thecoderscorner.com/products/apps/tcmenu-designer/

closed 2021-02-16 by davetcc