TcMenu/tcMenu #109: no scrollbars

msalajka, 2021-01-18:
hello,

when I run _Generate code_, I only see part of the window without scroolbars, I can't continue.

[image]

m.

**win:**
Microsoft Windows [Version 10.0.18363.1316]
**resolution:**
1920*1080 (recommended)
text, app, etc. size: 150% (recommended)
**java:**
openjdk version "15.0.1" 2020-10-20
OpenJDK Runtime Environment AdoptOpenJDK (build 15.0.1+9)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 15.0.1+9, mixed mode, sharing)
**tcMenu:**
tcMenuDesigner-1.7.0.tar.gz
**run:**
C:\Devel\Java\jdk15\bin\java --module-path ../deps --add-modules com.thecoderscorner.tcmenu.menuEditorUI com.thecoderscorner.menu.editorui.MenuEditorApp

davetcc, 2021-01-18:
Given you're on Windows 10, can you not use the Windows 10 UWP version, it had some stability issues a month or two back, but it is now quite stable. It's available either in the Windows store "tcMenu Designer" or for download directly from our site. 

https://www.thecoderscorner.com/products/apps/tcmenu-designer/

I'll add this fix to the JavaUI backlog, to be fixed.

davetcc, 2021-02-06:
Scheduled for release 2.0

closed 2021-02-06 by davetcc