TcMenu/tcMenu #124: Headless designer workflow

dogtopus, 2021-04-21:
For people who doesn't like checking in any generated source code into version control system (like me), it makes sense to have a "headless" mode that generates code only using command line and the designer project file. This can also be linked with e.g. PlatformIO or custom build system so everything menu related can be automatically up-to-date during build time. Maybe a "portable version" that can be included as a git submodule would also be helpful here (see e.g. [ant-javacard](https://github.com/martinpaljak/ant-javacard))

davetcc, 2021-04-22:
Yes, I think this would be quite easy to achieve, the Java code generator runs quickly and could either be used from a JAR file within a build system or using the current executable approach instead. The JAR approach is probably more portable for this purpose, especially for integration with CI environments.

The JAR approach would just require a modern JVM to be available, then it could be truly cross-platform.

We would just need to provide an additional main endpoint, that only loaded the code generator and took the design to use as a parameter.

Once tcMenu designer 2.0 is out I'll take a look at how this can be achieved, it should be quite simple to get an MVP out.

davetcc, 2021-04-22:
In 2.0 it will muddy the water of what is considered generated code slightly, as for example themes for graphical displays are generated once by designer, then it is the intention that they would be modified by the user until they meet requirements. But yes, for the regular plugins and the menu definition files, I see the need for this.

dogtopus, 2021-04-22:
> In 2.0 it will muddy the water of what is considered generated code slightly, as for example themes for graphical displays are generated once by designer, then it is the intention that they would be [...]

I personally will consider those as code templates as they are generated only once and modified by the devs manually. Similar to copying them over and modify. It might still make sense to include them in the CLI generator for the sake of completeness, but with a flag or an entire new subcommand to activate the generation.

davetcc, 2021-04-28:
I've started this work on master, but not yet tried the command output:

tcMenuDesigner verify -f abc.emf

This ensures that the EEPROM addresses do not overlap, and also checks the project can load.

tcMenuDesigner generate -f abc.emf

This will generate everything, including all referenced plugins from the designer.

davetcc, 2021-04-28:
The commands are here, more can very easily be added, and they can take full advantage of any aspect of the API or project loading infrastructure: 

https://github.com/davetcc/tcMenu/tree/master/tcMenuGenerator/src/main/java/com/thecoderscorner/menu/editorui/cli

BTW, in the other repo you said that there were some boards tcMenu designer did not support, please can you let us know the boards, we'll fix them up. It's known to work on most Arduino, ESP and mbed devices.

But you have to understand that working on a board is more than just code compiling, there needs to be support in IoAbstraction and TaskManagerIO for that board too.

dogtopus, 2021-04-28:
> The commands are here, more can very easily be added, and they can take full advantage of any aspect of the API or project loading infrastructure: [...]

It's more of an UI/UX issue apparently. Like Arduino SAMD actually means any Arduino compatible boards that use ARM processor. I think changing the label on the option should be enough (like `ARM-based Arduino-compatible board (e.g. Arduino SAMD)`). Maybe change the Arduino UNO/... to `AVR-based Arduino-compatible board` too since Teensy 2.x is a thing. ESP can be left as ESP since they are special enough.

Also probably add a notice somewhere about the official test environment so people know what is guaranteed to work and what is "untested" or "not as well tested".

davetcc, 2021-04-30:
> It's more of an UI/UX issue apparently. Like Arduino SAMD actually means any Arduino compatible boards that use ARM processor

Don't forget you were using the Java UI which had become somewhat behind in terms of UX to the macOS and Windows store versions, as of 2.1 Java will have most of the UX improvements from those versions, as it will be the only version. We've completely removed registration from the startup flow and made it optional for all.

Also, the CLI is starting to take shape for anyone wanting to integrate with CI, and when #126 is complete it should be pretty much a single executable on Windows, Linux, and macOS.

davetcc, 2021-04-30:
This is now in testing and all committed, I am now creating a basic schema for EMF files, which are essentially JSON files. Any designer from 2.0 onwards already pretty prints the EMF JSON on output. Along with some documentation, this will essentially complete this task.

This has the advantage that a user can switch between the designer and command line at any time, and for nearly any flow. There's probably a need to run the UI once to configure everything at the moment, although even this can be handled with extra commands without too much effort.

davetcc, 2021-05-01:
This is in internal testing now, and we'll produce a beta once we've tried it a few times ourselves and have a workflow together for it.

davetcc, 2021-05-07:
First version going into 2.1.1 release