TcMenu/tcMenu #158: Code Generator for completely custom rendering starting from scratch, issues

Netsplits, 2021-07-22:
When using Code Generator with the renderer "completely custom rendering starting from scratch", at the required field "Renderer Header File", it doesn't let to put the ".h" extensions (returns validation error), but generated code it has not the extension added in the "#include" sentence.

Also it includes "BaseRenderer.h", I don't know is a typo and must be "BaseRenderers.h" or is done this way so the user can add a "BaseRenderer.h" file. Not sure if this is by design.

davetcc, 2021-07-22:
Thanks for reporting this, that plugin is not used much so probably has been like this for some time.

Task: Add the .h to the include definition in the plugin.

davetcc, 2021-07-22:
2.2 is being tested now, this is a very minor plugin only fix, we'll add it to the release.

davetcc, 2021-07-23:
Will be in 2.2.x

closed 2021-07-23 by davetcc

AtmosphEng, 2023-06-07:
I've still noticed something similar with a Mega2560 build with  ILI9341 SPI LCD controller. If I initially use the Designer Code Generator to use the "AdafruitGFX Quick Start" and then afterwards remove the display via "No display required", then the resulting Code Generator build fails with the following errors.
In the src_menu.h file: 
1) #include "tcMenuAdaFruitGfx.h" remains and has to be commented-out;
2) three externs also need to be commented-out (gfx, gfxDrawable, renderer)
In the project_menu.h file: #include <BaseRenderers.h> becomes shortened to "BaseRenderer.h" and the "s" needs to be restored. This is similar behaviour to that reported above.
HTH