TcMenu/tcMenu #168: Allow adding comments before items

emptynick, 2021-10-03:
Hi,

thanks for your wonderful library. It really helps me a lot!

For me it would be really helpful to be able to add comments before menu-items as the amount of lines gets out of hand pretty fast.
For example I generated a RGBW item that extends the RGB(A) item.
This way, I could add a `// TODO: Change to RGBW` and my IDE would scream at me everytime i re-generated the code from IDE.

Another nice thing would be conditionally rendering of items based on conditions.
I know that won't be easy because of previous/next items.

Unfortunately I am not good enough in Java to help out here.

Thanks,

Christoph.

davetcc, 2021-10-03:
> Another nice thing would be conditionally rendering of items based on conditions.

It is implemented, you can set an item as visible or readonly at runtime. See the menu item documentation. https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/

Comments, we'll maybe look at in the future, but are you aware you can create menu items manually and add them into structure. See the dynamicMenuItems example, it adds menu items in at runtime. Also, each menu type documentation shows you how to add it into code manually: https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/#specific-menu-item-documentation

The idea is that if you want special types, you'd create them in code and then add them to the menu using `menuMgr.addMenuAfter(itemBefore, itemToAdd);`

davetcc, 2021-10-07:
One of the main reasons that a shared reference would run into issues is down to how the notifications work. It also doesn’t scale to other fields well. The multi delegate allows more possibilities. 

Ie when something is marked visible it is also considered changed.  It’s this changed flag that makes the item draw again. And realistically after setting the visible flag on items, then the menu needs to be marked as structurally changed too via the menu manager because the drawing order is altered.