TcMenu/tcMenu #178: Make menu visible / invisible on button press

hackintoshlover, 2021-11-16:
Hi Dave,
I was trying to implement the visibility parameter but unfortunately not able to get it through.
Basically i have modified the analogdfrobot code to make "onLED2" visible when "onLED1" == true and make "onLED2" invisible when "onLED1" == false
It only turns visible but does not turn back to invisible when selected
Also its not instant after selecting the parameter i need to go back to main an enter again
Kindly refer to the attached video for reference
https://drive.google.com/file/d/1d5FChufHFT5iFiripVTWxWNIBY4cnSMh/view?usp=drivesdk
Final Question: How do i make a menu visible on long press of a switch and turn it back to invisible on selecting "exi
[image]
t"

hackintoshlover, 2021-11-16:
Library versions all latest
Arduino version 1.8.13

davetcc, 2021-11-16:
Can you please ask questions in the forum? https://www.thecoderscorner.com/jforum/recentTopics/list.page

It keeps the noise down in here.

After changing the visibility of menu items, you need to tell the menu manager the structure has changed, see https://github.com/davetcc/tcMenuLib/blob/f690c27345e735c040e4b62c0590f3b82ab568fa/src/tcMenu.h#L340

Unless there is a bug, please close and if need be open a question on the forum. Even there, there's no guarantee of an answer, but I try and answer things on a best efforts basis if it doesn't take too much time.

davetcc, 2021-12-06:
There was an issue with the menuMgr structure change notification handler. This will be released in V2.2.5 of tcMenuLib due shortly: 

https://github.com/davetcc/tcMenuLib/issues/130

This release depends on 2.1.0 of IoAbstraction and will be due soon.

closed 2021-12-06 by davetcc