TcMenu/tcMenu #184: Install instructions arch

cvanelteren, 2022-02-19:
Hi,

I am trying to get this package to run on arch but I find the install instructions rather hard to understand. I have got it to compile using java17-opensdk but I am stuck generating the app. 
From the instructions I ran from inside `tcMenu/tcMenuGenerator/target/jfx/app` this command:

`java --module-path ../deps "-Dprism.lcdtext=false" --add-modules com.thecoderscorner.tcmenu.menuEditorUI com.thecoderscorner.menu.editorui.cli.TcMenuDesignerCmd`

however, this stalls and gets me no output. Is this supposed to happen? The installation instructions go on to bind the app to an image with `jpackage`. However, on arch this tools is not available from what I can tell. Could somebody point me in the correct direction?

Side note: I am not that experienced with java ^^

davetcc, 2022-02-21:
To be honest, I tend to build on either PopOS or Ubuntu. It sounds to me like something is going wrong at startup, but without more details, I cannot say. Could you check in the ~/.tcmenu directory and see if there is any logging there.

davetcc, 2022-02-21:
Also, if you're bulding for yourself, the jpackage should not matter that much, although in any properly functioning JDK install I've seen it has been present. Certainly on Windows, Ubuntu, popOS and macOS.

For your own use, I'd just take the contents of the app directory and put it into a suitable bin directory to run from there, making a shell script that runs the command.

davetcc, 2022-03-20:
Did you take a look in the ~/.tcmenu directory to get the logs for what went wrong? I'll leave it open a few more days.

cvanelteren, 2022-03-20:
I will close this for now and look into this in the future.

closed 2022-03-20 by cvanelteren