TcMenu/tcMenu #208: Feature request: Add RP2040 support (using pico-sdk)

DatanoiseTV, 2022-05-19:
I would like to request support for the Raspberry Pico (RP2040) using the pico-sdk.

davetcc, 2022-05-29:
At the moment we're trying to stabilise the library and make it work more reliably on the boards that it already supports; which is a really wide range of boards already.

One of the biggest issues for this would be that we've not really looked at that API to see what's available. For tcMenu to work properly, TaskManagerIO, SimpleCollections and IoAbstraction must work properly.

For task manager and the circular buffer, there would be one complexity to solve, which would be how to perform atomic operations. Not in itself trivial as it is a dual-core cortex-0 device that may not support load and store exclusive (LDREX, STREX) making CAS operations more difficult.

However, I'd imagine the most complex task would be mapping all the hardware operations in IoAbstraction; which given that we've not used that API before could take some time. If someone was able to "fill in the blanks" on how the device IO worked and do a bit of testing, I'd help put the basics in place; IE create the abstractions that needed implementing in terms of device calls and testing.

davetcc, 2022-05-29:
Encoders, matrix keyboards, buttons, EEPROMs, and most LCDs including i2c ones would just work if IoAbstraction worked. 

But the next question would be display libraries. At least one plugin would be needed.

For ethernet, we'd just write a simple adapter that allowed the current remote to support it properly.

davetcc, 2022-08-12:
As per discussion in the IoAbstraction issue https://github.com/davetcc/IoAbstraction/issues/162 the Raspberry PI Pico has an Arduino/mbed core that should already be compatible with tcMenu. Using this core would still allow you to use the underlying capabilities directly if needed. As discussed in the other issue, with the latest version of all the required libraries no changes should be needed to make it work.

I'll close this for now, and if anything comes up during the RPI2040 testing I'll raise specific issues for those.

DatanoiseTV, 2022-08-13:
This only adds support for the RP2040 using the Arduino IDE and the rp2040 core. It would be nice to see non-arduino pico-sdk support, as the arduino implementation still misses a lot of features of the pico-sdk.

davetcc, 2022-08-13:
But that doesn’t really matter, Arduino is only the wiring library and build. It’s an easy means to get started. You can still use any feature in the underlying API that you wish. With the mbed core you can use mbed, rtos, CMSIS and the underlying apis. Same with the other core.

Take for example I2c on mbed cores I’ve been playing with using asynchronous calls. On some boards I’m using LwIP instead of Ethernet all while using Arduino build.

davetcc, 2022-08-14:
Once the new core (arduino-pico) is available in platformIO, we'll make that core work with our libraries, this core is a thinner wrapper directly on top of the SDK, you can even use the SDK cmake build system with it if you prefer, and as with the other core, you can use any SDK function that you wish to use.

davetcc, 2023-10-24:
I'm going to reopen this. RPi 2040 has become one of my most used cores, and with the recent problems in CLion integration and platformIO I think I'm going to move back more toward native tool chains, even if I sometimes use Arduino as a library on the top of them.

For the Pico SDK, the following will be needed:

* An IoAbstraction interface that calls the GPIO functions and handles interrupts.
* A wire impl that handles calls into I2C libraries.
* An SPI wrapper implementation.
* Task Manager may need a couple of tweaks, but its atomic support is already based on Pico SDK.

--- 

Similarly, for ESP32 IDF, Most of IoAbstraction on ESP32 uses IDF, I think an outside Arduino version would be quite trivial.

Lastly, for STM32, I think there would be some value in an STM32Cube version, especially now they are heading in the cmake direction as well.

davetcc, 2023-10-24:
In terms of displays, I'll see what's available once I have the core working, in the worst case that I can't find a suitable SSD1306 or TFT driver, I'll repurpose the mbed port of AdafruitGFX to also work for this core as well. Hopefully, that won't be needed.

Step 0 is to set up the native toolchain so I can get it working in CLion.

In case you're wondering which Pico PR / PlatformIO issues I'm talking about it is here: https://github.com/platformio/platform-raspberrypi/pull/36

davetcc, 2023-10-27:
I have a local build with SimpleCollections, TaskManagerIO and IoAbstraction working from pico-sdk cmake. The rest should be much easier from this point in. I do not jest when I say this could beta in 4.2. Closely followed by STM32Cube and ESP32-IDF.

davetcc, 2023-11-18:
PicoSDK will be available in 4.2. ESP-IDF and Stm32Cube will be possible but will come later when the UI and mobile app are more stable.

closed 2023-11-18 by davetcc