TcMenu/tcMenu #219: ESP32 and IL

simplo, 2022-09-02:
Hi guys, just  using TCMenu for first time.

I generated a code for ESP32, serial control and ADAFruit GFX for IL9341.

Once I compile it I get this error:

**************************************************
C:\Users\simplo\Documents\Arduino\libraries\IoAbstraction\src\esp32\ESP32AnalogDevice.cpp: In member function 'uint16_t EspAnalogInputMode::getCurrentReading()':
C:\Users\simplo\Documents\Arduino\libraries\IoAbstraction\src\esp32\ESP32AnalogDevice.cpp:67:15: error: 'DAC1' was not declared in this scope
     if(pin == DAC1 || pin == DAC2) {
               ^~~~
C:\Users\simplo\Documents\Arduino\libraries\IoAbstraction\src\esp32\ESP32AnalogDevice.cpp:67:15: note: suggested alternative: 'D11'
     if(pin == DAC1 || pin == DAC2) {
               ^~~~
               D11
C:\Users\simplo\Documents\Arduino\libraries\IoAbstraction\src\esp32\ESP32AnalogDevice.cpp:67:30: error: 'DAC2' was not declared in this scope
     if(pin == DAC1 || pin == DAC2) {
                              ^~~~
C:\Users\simplo\Documents\Arduino\libraries\IoAbstraction\src\esp32\ESP32AnalogDevice.cpp:67:30: note: suggested alternative: 'D12'
     if(pin == DAC1 || pin == DAC2) {
                              ^~~~
                              D12
exit status 1
Errore durante la compilazione per la scheda WEMOS D1 R32.

**************************************************

Please note that my ESP board is a Wemos DI R32

davetcc, 2022-09-07:
Can you find out exactly which device is onboard, I suspect it was an ESP32S3 as the DAC was removed. If that is the case, the latest IoAbstraction would solve this problem - [2.3.5](https://github.com/davetcc/IoAbstraction/tree/2.3.5). Make sure to update TaskManagerIO and SimpleCollections at the same time.

simplo, 2022-09-07:
Hi Dave, I'll try the 2.3.5 as soon as possible.
At the moment I fixed changing line 66 on ESP32AnalogDevice like this:
	#ifdef ESP_HAS_DAC
			if(pin == D11 || pin == D12) {
			     dac_output_disable(pin == D11 ? DAC_CHANNEL_1 : DAC_CHANNEL_2);
		    }

davetcc, 2022-09-21:
Can we close this? Did upgrading to the newer IoAbstraction work?

simplo, 2022-09-21:
Sorry Dave, I'm totally focused on another project and didn't had the time to check your solution.
I really don't know when I will do it :(

davetcc, 2022-09-21:
No problem I think it’s fixed by the new IoAbstraction version so I’ll go ahead and close it. If it doesn’t fix it when you upgrade please reopen

closed 2022-09-21 by davetcc