TcMenu/tcMenu #221: Add optional "step" attribute to the AnalogMenuItem

vzahradnik, 2022-09-24:
[image]

`AnalogMenuItem` allows to define minimum and maximum value as well as other useful features but it is missing an option to specify increment/decrement step.

In most cases, it's fine. We want to increment by +1. But in some cases it is useful to increment or decrement the value in bigger steps. For example, display contrast. If we change the value by 1, user hardly notices a change. In this case, I'd like to define a step telling TcMenu how much to change the value.

It would be great to have direct support in TcMenu Designer with increment step being optional and defaulting to 1.

As a workaround, I was thinking to implement the change in callback function. However, once the value was changed, I can't tell whether it was incremented or decremented.

Thanks for the feedback!

davetcc, 2022-09-24:
Agreed, this case has never really been thought of so far, but is a very valid case, I think either an optional step on the end of the analog info block or a way of setting it on the object later would work. I'm planning on a release soon anyway so it could go in that, just got to work out how to make it backward compatible.

davetcc, 2022-10-04:
releasing within the next day, going through final testing.

closed 2022-10-04 by davetcc