TcMenu/tcMenu #222: Allow users to customize timeout value after which the menu is reset to root

vzahradnik, 2022-09-24:
When using TcMenu library, I noticed that once I go into a sub-menu and don't touch the display for 30 seconds, the menu hierarchy is reset back to root (top menu).

I have implemented some menu screens with status information which is periodically refreshed. While looking at the display we hardly press any buttons. For usecases like mine, it would be great if TcMenu allowed us to configure custom timeout value or turn this feature completely off.

[image]

davetcc, 2022-09-24:
Take a look at http://thecoderscorner.com/products/arduino-libraries/tc-menu/renderer-take-over-display/#functional-approach-to-capturing-display-timeout--reset as that may do what you want.

vzahradnik, 2022-09-24:
Thanks, I will have a look.

davetcc, 2022-09-24:
The rendering of the boolean items seems a bit misaligned there, I'll take a look at many boolean items on the same page and see if any tidying up is needed.

vzahradnik, 2022-09-24:
Might be but this is not much of an issue for me. I still work on a prototype so this is good enough. I have some bigger features in mind. Once I have something concrete, I will create separate issues and we can discuss there.

Your advice helped. I dig through the `MenuManager` code as I thought the logic was there. I should check the renderer, too :)

Thanks for your help!

closed 2022-09-24 by vzahradnik