TcMenu/tcMenu #227: UI for "default value"

luwenjin, 2022-10-09:
I've noticed there is a  "defaultVal" param in MenuItem class definition (eg: AnalogMenuItem).
But in the Designer UI, there is no such option.

It would be really nice to have "default value" option on the Designer UI, 
so that many tricky code for handling default values can be deleted.

davetcc, 2022-10-09:
Theoretically, it should not be too difficult, but I'll need to think about how to present this in the UI and also at what point the defaults should be applied, probably during the setup menu call, but that would mean anyone loading back from EEPROM would need to do so afterward.

luwenjin, 2022-10-10:
After further digging into code, it seems that, defaultVal is only used before loading value from EEPROM.
But I guess currently there is no easy way to tell if the value loaded from EEPROM is UNINITIALIZED or ZERO, because both are equal to 0x0000.

davetcc, 2022-10-10:
Yep, actually, I've looked at this again, and we are talking about setting it in the constructor, which is before any other operation can be performed, so it is completely safe. We already know what kind of value can be stored in the field, thanks to the underlying Java API, so I just need to think about how to present it on the UI now.

luwenjin, 2022-10-10:
I realized I might misunderstand the purpose of the `defaultVal` param, I thought it was the default value to store in EEPROM if it has not been changed.

Here is my use case:
1. the firmware had a bunch of settings as MenuItems stored in EEPROM
2. I had initialized all the settings with default values on the first launch (no problem so far).
3. Later, I added a new MenuItem with EEPROM addr to my code, and updated the MCU.

The problem is that the new setting is Volume, which should not be 0 by default, but I allow user to set it to 0.
Can I set it's default value to like 5 with the future DesignerUI ?
Or is there a way to tell if the MenuItem(with EEPROM addr) is initialized, so that I can set it's default value by code.

davetcc, 2022-10-10:
1 and 2 would be handled no problem by the new default value case, but 3 would not. I think the default value is a good idea to handle the initial setup, and easy to add.

The way that EEPROM migration from one version to another works in tcMenu is that you would change the magic key, IE the two bytes at the start of the ROM it compares against. Then inside the new code logic, if the old magic key was found, you'd put the new settings into the EEPROM, change the magic key to the new one, and run the load again. For example in pseudo-code

    Attempt menu load
    IF Menu load fails because the magic key does not match
       IF magic key matches the previous magic key - upgraded device case
           Write the differences between versions to the ROM and commit (maybe from a constant array)
           Change the magic key (first two bytes of ROM) to the new one
           Run the load again
       ELSE
            Do whatever is needed for a new device
       END
    END

luwenjin, 2022-10-10:
Using magic key to track migrations is brilliant! 👍🏻
Looking forward to the new designer. 😄

luwenjin, 2022-10-11:
I guess in most cases, only field addition (EEPROM addr +n) will happen in migrations. 
So Is it possible to track the `max saved EEPROM addr` by library, and handle the default value with these pseudo code:
```
// load from EEPROM
FOR every menuitem with EEPROM addr
   IF menuitem's eeprom_addr <= max saved EEPROM addr:
        load value from EEPROM to menuitem
   ELSE
       do nothing (meaning menuitem.value = menuitem.defaultVal )
  END
ENDFOR
```

When save to EEPROM, update the "max written EEPROM addr" to the largest EEPROM addr.

With this method, most migrations cases can be covered by library automatically.

davetcc, 2022-10-15:
See https://github.com/davetcc/tcMenuLib/issues/157 for the library side of this, in-flight commit supports everything apart from text items, they will come next.

davetcc, 2022-10-23:
All types that have values now support default value.

closed 2022-10-25 by davetcc