TcMenu/tcMenu #234: bug: renderer.turnOffResetLogic() not working

luwenjin, 2022-10-28:
tcMenu version
---
2.5.0 （2.4.x can also reproduce）

How to reproduce:
---
1. open example simpleU8g2
2. add `renderer.turnOffResetLogic();` before `setupMenu()`
3. build, upload, enter menu `settings`
4. wait 30s
5. menu will return to root 

But it should never leave the submenu automatically because of `renderer.turnOffResetLogic();`

davetcc, 2022-10-28:
It probably needs to be made clearer in the documentation but any methods on menu manager or renderer should only be called after setupMenu, as the object is uninitiated before that. But I’ll see if it is possible to handle this during initialisation.

luwenjin, 2022-10-28:
I see, I'll try to reorder the code on Monday. 😄

luwenjin, 2022-10-31:
`renderer.turnOffResetLogic();` after `setupMenu()` works!
If the order is important and impossible to change, maybe add a warning or raise an exception ?

davetcc, 2022-10-31:
Exceptions are turned off on most boards, even for operator new so I don't think that would work. Maybe we could add a logger warning, indicating "Menu System is not yet initialized" but how many people use IoLogging, not really that sure. I will make sure it's properly documented though

davetcc, 2022-11-05:
I will add an IoLogging warning if any set method is called before init. I think it's the best we can do.

davetcc, 2022-11-05:
See https://github.com/davetcc/tcMenuLib/issues/165

closed 2022-12-17 by davetcc