TcMenu/tcMenu #239: Feature Request: improve Key Repeat Speed

luwenjin, 2022-10-28:
I've noticed Designer Add a "Key Repeat Speed" option, It is great, before that I was thinking to tweak this value by copying the library and modify the code. 

The value range is between 5~50, when I changed it to 5, I can change analog value much faster with buttons.

However, to my feeling, even the fastest speed is still slow in some case.
For example, It took 15 seconds to change the analog value from 0~255, and the max analog range has 5 digits, makes large analog menuItem almost unsable.

So how about change "Key Repeat Speed" min value and expose more options like "acceleration" / "max speed" …… ?

davetcc, 2022-10-28:
Sure I’ll lower the minimum to 1 and also check that acceleration is enough too.

davetcc, 2022-10-29:
BTW, even without needing to copy. You can always customize input just by setting the code generator in designer to "no input required" and then write the input with any options you like. http://thecoderscorner.com/products/arduino-libraries/tc-menu/menumanager-and-iteration/#controlling-the-menu-items-manually

luwenjin, 2022-10-29:
Nice to know that 👍

davetcc, 2022-10-29:
For ESP touch buttons - 1 to 50
For dfRobot shield - 1 to 50
For up-down encoder - 1 to 50
For analog joystick - the delay and divisor are exposed.
For matrix keyboards - the delay and repeat were already exposed.

I think that covers all the possibilities.

Also, you can adjust the default poll interval for all switches, which defaults to 20mills by setting the build flag SWITCH_POLL_INTERVAL to another value, for example, to poll 100 times/sec set to 10.

luwenjin, 2022-10-29:
Great!

davetcc, 2022-10-30:
Along with more configurability coming in IoAbstraction 2.5.0 (yet to be documented) this will allow a greater degree of control over acceleration and repeat key speed.

closed 2022-12-17 by davetcc