TcMenu/tcMenu #241: Suggesting: make dialog's header/bodytext unselectable

luwenjin, 2022-11-03:
I've tried dialog function recently. 
My purpose was to show a loading dialog, and it would be dismissed by code.

However the dialog's header/bodytext are selectable, 
if select header, it will go to root menu,
if select body text, nothing will happen (but the cursur is focused on the body text).

So it feels a bit odd, it might be better make header/bodytext unselectable .

davetcc, 2022-11-05:
The header is selectable and is actually treated as a valid exit point from the menu (the equivalent of the close button by default IIRC), you can disable it from taking action, dynamicMenuItems example shows a few examples of how to do that. 

However, at the moment there isn't an easy way to make something unselectable. I'm not sure how we'd do it and it would need a lot of thought. You can at least control what it does though with either a `BaseDialogController` or commit callback / `CommitCallbackObserver`.

By default the first dialog button is made active when UI is shown, but dialogs are actually full menus in their own right, and they can contain other menu items,

davetcc, 2022-11-05:
Actually, if I implemented the active item callback - https://github.com/davetcc/tcMenuLib/issues/164, then you could probably just avoid selecting the dialog item's text field, If you look at `MenuBasedDialog` you can get the text field using getBufferMenuItem(), you could just stop it being activated, and move activation to it's getNext()

luwenjin, 2022-11-05:
I will do some experiments on Moneday. 😆

luwenjin, 2022-11-07:
I've no question about this issue now, just waiting for the new version 😄

luwenjin, 2022-11-07:
I feel it's necessary to provide a way to change the header's selectability, here are at least 3 use cases: 
1、Don't want the root menu's header to be selected, and don't want to provide a "About Info" for the header.
2、Want to "Lock" user under certain Submenu, and only switch menu tree by code.
3、Want to make the dialog mandatory (eg: a loading dialog)

davetcc, 2022-11-07:
Yep, as of 3.0 you'll be able to get notifications when the selection changes, you could effectively forbid a selection by reselecting a different item. There's also the dialog controller route that can completely override nearly all dialog operations.

davetcc, 2022-11-07:
I'll close this as fixed by #244

closed 2022-11-07 by davetcc