TcMenu/tcMenu #242: Bug: Refresh ListRuntimeMenuItem inplace will cause additional items unable to nav to

luwenjin, 2022-11-03:
I've create a ListRuntimeMenuItem, and use the modified demo code from the website
```
int CALLBACK_FUNCTION fnNewRuntimeListRtCall(RuntimeMenuItem * item, uint8_t row, 
                                                  RenderFnMode mode, 
                                                  char * buffer, int bufferSize) {
    switch(mode) {
        case RENDERFN_INVOKE:
            item->setNumberOfRows(item->getNumberOfRows() + 1);
            // menuMgr.navigateToMenu(item);
            return true;
        case RENDERFN_NAME:
            strcpy(buffer, "Name");
            fastltoa(buffer, row, 3, NOT_PADDED, bufferSize);
            return true;
        case RENDERFN_VALUE:
            strcpy(buffer, "Val");
            fastltoa(buffer, row, 3, NOT_PADDED, bufferSize);
            return true;
        case RENDERFN_EEPROM_POS: return 0xFFFF;
        default: return false;
    }
}
```

In my code, I changed NumberOfRows in RENDERFN_INVOKE.
After this, the addition row appeared, but **I can not nav down to the new item**.

So I tried to workaround by adding `menuMgr.navigateToMenu(item)` below `setNumberOfRows`. 
It solved the navigation problem, but now **I need to select the header twice to go back to parent menu**.

luwenjin, 2022-11-05:
The actual use case is:
I made the last row of the list a "Refesh" button, once it's pressed, the list will grow/shrink according to new data.

davetcc, 2022-11-05:
Navigating again will not do what you want, because it will push the list onto the navigation stack twice. I've never tried changing the number of items in a list while it is displayed. My first thought is does `menuMgr.notifyStructureChanged()` work.

That would force the whole UI to get re-evaluated.

luwenjin, 2022-11-05:
Thanks for the advice, I'll try that on Monday.

luwenjin, 2022-11-07:
`menuMgr.notifyStructureChanged()` has no effect. 😢 

My current workaround is `navigateToMenu` to root first, then `navigateToMenu` to the current list, these happens before rendering so users will not able to notice.

And also, if I navigate to the current list menu after the list growth, despite that I have to  activate the header twice to go back to parent, I can freely navigate the list up and down.

davetcc, 2022-11-07:
I'll look at this, there must be something amiss somewhere as structure changed should have fixed it.

davetcc, 2022-12-17:
I've been thinking about this, I think we should handle this similar to how it would probably be handled in a desktop library, Something like this would probably just be pushed onto the UI queue in MacOS, Windows or JavaFX.  Can you try to run a taskmanager task to do it later so we are not changing the list while it's being drawn?

```
void myTaskManagerTaskFn() {
   myListItem.setNumberOfRows(item->getNumberOfRows() + 1);
}

// then later on in the switch statement
        case RENDERFN_INVOKE:
            taskManager.execute(myTaskManagerTaskFn);
            return true;
```

luwenjin, 2022-12-20:
Set number of rows in a task has no effect.
Still unable to navigate down to the additional row.

davetcc, 2022-12-22:
Ok thanks. I’ll debug and feed back

davetcc, 2022-12-31:
3.0 will add `menuMgr.recalculateListIfOnDisplay(item);` which forces the encoder if present to refresh, it wasn't the renderer, it was just that the encoder max range had not been increased. The `examples/arduino32/piPicoTftEncoder` has been modified to make use of it.

davetcc, 2022-12-31:
Committed to wrong issue number, but the bug is fixed.

closed 2022-12-31 by davetcc