TcMenu/tcMenu #243: Suggestion: add onChange Function to RuntimeListMenuItem

luwenjin, 2022-11-04:
It seems that the onChange Function is disabled on the designer UI.
Is it possible to add onChange callback to it, so that it can be consistent to other menu items (eg. RuntimeListMenuItem's use case is very close to EnumMenuItem, we tend to change between these two menu item types).

davetcc, 2022-11-05:
Hi there, it's disabled for lists because the list has an invoke handler within the callback function that does the same thing. Lists work fundamentally differently in that the whole runtime callback is handled in user code, so there's a contract that you need to handle the runtime list callback.

It's worth taking a look through the documentation on runtime menu items and list menu items in the UI:

https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/based-on-runtimemenuitem/

and

https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/list-menu-item/

Given this, I don't think there's a much better way to handle this here.

luwenjin, 2022-11-05:
I see, you are right, I can use the invoke handler instead.

closed 2022-11-05 by luwenjin