TcMenu/tcMenu #244: Suggestion: add something like "onNav(MenuItem *item)" to MenuManagerObserver

luwenjin, 2022-11-04:
So that it's easier to get notified when a menu item is been focused.

Here is my use case:
I have some menu items for my slave devices, eg:
```
> Slave 1: MAC_ADDR1
  Slave 2: MAC_ADDR2
```
When the cursor is on the slave's menuitem, I would like to flash that slave's LED to indicate which one is which.

davetcc, 2022-11-05:
Yes, this has come up a few times now, in slightly different ways, but I think it is time to find a way to handle this correctly, it will help a few people.

It's got a little complexity because matrix keyboards, rotary encoders, and touch screens use slightly different logic.

davetcc, 2022-11-05:
See https://github.com/davetcc/tcMenuLib/issues/164

davetcc, 2022-11-05:
Actually, this turned out to be trival. See the commit on the linked issue

closed 2022-12-17 by davetcc