TcMenu/tcMenu #246: Make at least thecoderscorner products website opensource

davetcc, 2022-11-05:
This has come up a few times now, and people would like to be able to contribute to the repo. I see no reason not to make at least the products section (containing the library and app docs) open source, just need to give a bit of thought to the mechanics of that and how it would work. In the simplest case, it would just need a run of Hugo on the server once merged.

vzahradnik, 2023-02-16:
Just ping me if you need any help with that. I used to run my blog site via GitHub pages which was automatically built using GitHub Actions.

As for inspiration, Home Assistant guys have a quite good system.

- There is a stable branch for the current release. In this branch, only critical changes are accepted.
- Then there is a branch called "Next". There users contribute their changes for the upcoming release. For example, when there is a new attribute in EEPROM management (as you introduced recently), this documentation change gets released with the library.
- As part of the release, the Next branch is merged into main branch. This triggers GitHub Actions and new documentation is online.

A very good first step would be to separate your documentation into a separate repository.

davetcc, 2023-03-01:
So at the moment, the plan of action is:

* Finish off the card layout and dashboard support.
* Get the UI i18n to at least a reasonable place where most common dialogs are done.
* Put the i18n support into the code generator
* Create a BETA version of the UI and ask a few people to run with that for a week or two, make sure it's stable
--
* Take the products part of the site and move it to GitHub.
* In the first instance just link the GitHub repo from the private site repo so it pulls master and publishes.

vzahradnik, 2023-03-01:
I work with your nightly code, so count me in. By the way, was there any significant change in TagVal protocol? I'm asking because of the Python version I work on.

davetcc, 2023-03-01:
I think the only recent change is to send checkbox for Boolean items.

davetcc, 2023-05-12:
Phase 1 - https://github.com/davetcc/arduino-libraries the libraries are separated out from the main site and pulled in using a git submodule in the private build.

davetcc, 2023-05-12:
Phase 2 will be a minimal hugo project that can allow for running up the site and looking at it before committing.

davetcc, 2023-05-12:
A new build of the webserver (was overdue for a while) http://betatest.thecoderscorner.com/  is running, it is built from the hybrid site.

davetcc, 2023-05-13:
The site based on the open source arduino-libraries repo below is now live, there are a few issues to iron out here.

At the moment I have literally just broken out a part of the larger Hugo site, which means that the images and theme areas are not available.  However, this at least allows people to start editing the documentation, and in a second iteration, if a lot of people start editing, we can discuss how to fully separate that documentation into a complete Hugo build or some other build that is entirely self-contained.

Repo containing all Arduino library documentation: https://github.com/davetcc/arduino-libraries

closed 2023-05-13 by davetcc