TcMenu/tcMenu #254: BUG: after renderer.giveBackDisplay() + menuMgr.activateMenuItem(&secondItem), moving down twice required to really move the cursor down

luwenjin, 2022-11-13:
There might be 2 bugs: 

1. I selected my second menuItem under a submenu, after `renderer.takeOverDisplay();` and `renderer.giveBackDisplay()` in the callback, I found the cursor is on the first child item (which should be on the 2nd item like before taking over display).
2. Due to bug 1, I ran `menuMgr.activateMenuItem(&secondItem)` after `renderer.takeOverDisplay()`, the cursor position is correct. But when I press DOWN for the first time, the cursor won't move down, after pressing for the 2nd time, the cursor moved and working smoothly afterwards. (if I press UP at first, no problem at all).

davetcc, 2022-11-13:
At the moment, giving back the display does a reset of the current menu. In many menu designs, this occurs when there's a reset timeout and the user UI takes over control. 

Point 2 should be fixed in 3.0.0 because of another issue that was reported in the forum.

davetcc, 2022-11-13:
This should be fixed in 3.0.

closed 2022-12-17 by davetcc