TcMenu/tcMenu #257: Can SubMenuItem's name be changed in runtime ?

luwenjin, 2022-11-20:
I have a SubMenuItem contains several boolean items, make it be used as a multi select form field.
I would like the SubMenuItem's name reflect the values of the inner boolean items, so users don't have to enter the menu to see what have been checked before.

I've looked up the code, it seems `menu.getNamePGMUnsafe()` is the closest to what I want.
But it returns a const pointer, so I'm not able to strcpy another chars to it.

Are there other ways to change the SubMenuItem's name?

davetcc, 2022-11-21:
In summary, yes, but the way the designer creates the item doesn't make this easy to do. I've already got a plan for 3.1 to sort this out by allowing any runtime items to have a custom callback (similar to how lists work now). I'm in the middle of sorting out Unicode support at the moment so I don't want to get too deep into this right now. 

But in short, yes it is possible now, but the designer doesn't make it easy to do (but that can be changed very easily to improve matters.

davetcc, 2023-01-07:
If it is for a fully dynamic menu, you can just create the items at runtime and that works now. The dynamic menu example shows how to do this. The full solution will come in 3.1 or 3.2 when you can choose to customize the callback of runtime items on a per-item basis.

davetcc, 2023-01-07:
Provide within the designer UI:

* For all regular menu items (value types, float etc) provide the ability to create an info struct that is in RAM, so it can be altered later. Checkbox would be "Create item in RAM (allows modification)"
* For all runtime items, provide the ability to override the callback, so that you can both alter the behavior as needed (eg provide a filtered text control), by having an additional checkbox "Create custom runtime callback" which when checked would create the runtime item with a custom callback that always called back to the parent one.

davetcc, 2023-01-07:
also see #247 this is somewhat an extension of that.

davetcc, 2023-01-07:
https://github.com/davetcc/tcMenuLib/issues/166

closed 2023-01-22 by davetcc