TcMenu/tcMenu #258: Suggestion: use id(instead of index) to store EnumMenuItem

luwenjin, 2022-11-20:
I have an EnumMenuItem which contains several time intervals, eg: 10s, 30s, 1m，……

Currently, the value stored in EEPROM is the index of the row.
But as the program iterate, the rows may be added/removed, make the index point to a mismatched value.

If we can specify an id to each value, and store the id in EEPROM, there will be no need to worry about the values rearrangement.
For compatibility and usability considerations, id can default to the index.

davetcc, 2022-11-21:
Enums are really for the simplest of cases, if you need this level of complexity, why not use a scroll-choice menu item, you get full control of how it renders and everything? 

See https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/scrollchoice-menu-item/

luwenjin, 2022-11-22:
I see. Never thought about using ScrollChoiceMenuItem before,  suits my needs.👍

davetcc, 2022-11-29:
I'll close this for now, but if you think there's anything else, please re-open

closed 2022-11-29 by davetcc