TcMenu/tcMenu #268: Can "Navigate to Parent" focus the entrance item?

luwenjin, 2022-12-20:
for example, I have a menu structure like this: 
```
A
B
- B1
C
- C1
D
- D1
```

When I navigated to B1, if I go to parent using the title, the focus will be on the first child of the parent menu (A).

I think it would be better if its focused on B, since it's the entrance of the submenu.
And if we have consecutive options to set (B, C, D), by focusing on the entrance menu, the navigation experience will be more fluent.

davetcc, 2023-01-02:
On using the back menu, the navigator should put you back into the menu at whatever point you were before.

For example, if you clicked on B it would initially select the back menu item for B, which is the title.
Then should you click on that back menu item, it will pop the navigation one level, and that should select B again.

I'm not sure I fully understand the problem to be honest, as it is now, unless something has gone wrong, navigating backward using the back menu item, the previously selected value will always become active. Normally that would be the submenu you clicked on. However, clicking on a submenu (such as B above) will result in the back menu becoming active, is this what the issue is for?

luwenjin, 2023-01-05:
> On using the back menu, the navigator should put you back into the menu at whatever point you were before.

Yep, that's what I was expecting.

However, I've tried with simpleU8G2 example with 3.0.0 lib in these steps:
1. click settings -> enter settings and the title is selected ✅ 
2. click the title -> go back to root and selected "Toaster"(the first item) instead of "settings"  💔 

It's the same as in my app.

davetcc, 2023-01-05:
OK that is very odd, have you adjusted the number of navigation levels or something like that?

The default behaviour when clicking on the back menu item (title of sub-menu) is to go back to whatever was selected before, IE whatever was the active item when the navigate function was called.

luwenjin, 2023-01-05:
It's the first level menu under root, I didn't go any deeper, and never touched MAX_MENU_DEPTH before.
Can you reproduce this issue with example `esp32/simpleU8G2` ?

davetcc, 2023-01-05:
This is something that must have either gone wrong very close to the end of the dev cycle for 3.0 or not show on all environments, as it honestly did work during testing, but as of the release, agreed it is also broken for me. I'll debug and hope to patch the library soon. Thanks for reporting this.

davetcc, 2023-01-05:
https://github.com/davetcc/tcMenuLib/issues/175

luwenjin, 2023-01-06:
This bug also exists before I update to 3.0.0, you may check the code before 3.0.0

closed 2023-01-06 by davetcc