TcMenu/tcMenu #269: Build error

khvalera, 2022-12-21:
archlinux
packages installed:
sudo pacman -S maven java17-openjfx java17-openjfx-src jdk17-openjdk jre17-openjdk jre17-openjdk-headless
```
git clone https://github.com/davetcc/tcMenu.git
cd tcMenu
git checkout 2.5.0
cd tcMenuGenerator
```

[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /home/user/tcMenu/embedCONTROLFx/src/main/java/module-info.java:[2,20] module not found: javafx.fxml
[ERROR] /home/user/tcMenu/embedCONTROLFx/src/main/java/module-info.java:[3,20] module not found: javafx.graphics
[ERROR] /home/user/tcMenu/embedCONTROLFx/src/main/java/module-info.java:[4,20] module not found: javafx.base
[ERROR] /home/user/tcMenu/embedCONTROLFx/src/main/java/module-info.java:[5,20] module not found: javafx.controls

davetcc, 2022-12-21:
You need have javafx packages available. I normally use the Liberica full JDK that includes Java fx. You can either do that, add javafx yourself or add as a maven dependency

davetcc, 2022-12-22:
As I said before, the easiest way by far is to use Liberica Full JDK which has a build for your platform. It includes everything that you need. The second best way, is to just add the JavaFX dependencies yourself in the maven pom.

In the tcMenuGenerator pom.xml you'll see all the javafx dependencies there, but they are at scope "test", if you were to comment out the scope on those, it would also work with most other open JDKs.

For embed control they are literally commented out, just uncomment them - search for `  ` in the embedCONTROLfx pom.xml

davetcc, 2022-12-22:
Let me know if neither of those work, happy to help get this working on that platform.

khvalera, 2022-12-22:
Unfortunately, I'm not an expert in java, maybe I can't understand something :(
Commented out:
```
        <!-- javafx - when not using bellsoft or manually including Open JFX include the below -->

        <dependency>
            <groupId>org.openjfx</groupId>
            <artifactId>javafx-base</artifactId>
            <version>${jfx.version}</version>
            <!--scope>test</scope-->
        </dependency>
        <dependency>
            <groupId>org.openjfx</groupId>
            <artifactId>javafx-controls</artifactId>
            <version>${jfx.version}</version>
            <!--scope>test</scope-->
        </dependency>
        <dependency>
            <groupId>org.openjfx</groupId>
            <artifactId>javafx-fxml</artifactId>
            <version>${jfx.version}</version>
            <!--scope>test</scope-->
        </dependency>
        <dependency>
            <groupId>org.openjfx</groupId>
            <artifactId>javafx-graphics</artifactId>
            <version>${jfx.version}</version>
            <!--scope>test</scope-->
        </dependency>
```
`mvn clean install`
I get an error:
```
[INFO] -------------------------------------------------------
[INFO]  T E S T S
[INFO] -------------------------------------------------------
[INFO] 
[INFO] Results:
[INFO] 
[INFO] Tests run: 0, Failures: 0, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  5.867 s
[INFO] Finished at: 2022-12-22T13:19:02+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M3:test (default-test) on project menuEditorUI: Execution default-test of goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M3:test failed: Unsupported class file major version 61 -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginExecutionException
```
`mvn clean install -DskipTests`
build succeeds
is it ok to use the -DskipTests option?

khvalera, 2022-12-22:
the program is running, the menu is being created, but during generation it is not possible to select another display option
`java --module-path target/jfx/deps "-Dprism.lcdtext=false" --add-modules com.thecoderscorner.tcmenu.menuEditorUI com.thecoderscorner.menu.editorui.cli.TcMenuDesignerCmd $@`

[image]

davetcc, 2022-12-22:
Also, what happens when you click on the image button with "change" underneath, does it silently fail to show the popup list of possible other plugins?

khvalera, 2022-12-22:
The contents of the ~/.tcmenu directory are created:
```
.tcmenu
├── logs
│   └── tcmenu.log.0
└── plugins
    ├── core-display
    │   ├── adaArduinoSimpleBuilder.xml
    │   ├── adaGfxDisplay.xml
    │   ├── adaGfxDriver
    │   │   ├── tcMenuAdaFruitGfx.cpp
    │   │   └── tcMenuAdaFruitGfx.h
    │   ├── adaMbedOled.xml
    │   ├── adaTouchDriver
    │   │   └── tcMenuAdaTouchDriver.h
    │   ├── analogJoystickInput.xml
    │   ├── analogResistiveTouchScreen.xml
    │   ├── ArduinoTouchLibraryIntegration.xml
    │   ├── customDisplay.xml
    │   ├── customDrawable.xml
    │   ├── dfRobotJoystick.xml
    │   ├── dfRobotLcd.xml
    │   ├── EmbeddedJFX.xml
    │   ├── emb_jfx
    │   │   ├── JfxLocalAutoUI.java
    │   │   └── LocalTreeComponentManager.java
    │   ├── emptyDrawable
    │   │   └── CustomDisplayDrawable.h
    │   ├── esp32Touch
    │   │   ├── ESP32TouchKeysAbstraction.cpp
    │   │   └── ESP32TouchKeysAbstraction.h
    │   ├── esp32TouchSensor.xml
    │   ├── fullLcd
    │   │   ├── tcMenuLiquidCrystal.cpp
    │   │   └── tcMenuLiquidCrystal.h
    │   ├── i2cLiquidCrystalDisplay.xml
    │   ├── Images
    │   │   ├── adagfx-color.jpg
    │   │   ├── custom-display.jpg
    │   │   ├── DfRobotShield.jpg
    │   │   ├── embedded-java.png
    │   │   ├── esp8266.jpg
    │   │   ├── joystick.jpg
    │   │   ├── lcd-display.jpg
    │   │   ├── matrix-keyboard.jpg
    │   │   ├── no-display.png
    │   │   ├── oled-display.jpg
    │   │   ├── resistive-touch.jpg
    │   │   ├── rotary-encoder.jpg
    │   │   ├── ssd1306ascii.jpg
    │   │   ├── touch-pad-sensor.jpg
    │   │   └── up-down-encoder.jpg
    │   ├── liquidCrystalDisplay.xml
    │   ├── matrixKeyboard.xml
    │   ├── mbedStm32Bsp
    │   │   ├── BspUserSettings.h
    │   │   ├── tcMenuStChromaArt.cpp
    │   │   └── tcMenuStChromaArt.h
    │   ├── noDisplay.xml
    │   ├── noInput.xml
    │   ├── rotaryEncoderInput.xml
    │   ├── ssd1306asciiDisplayQuickStart.xml
    │   ├── ssd1306asciiDisplay.xml
    │   ├── ssd1306AsciiSupport
    │   │   ├── ssd1306asciiRenderer.cpp
    │   │   └── ssd1306asciiRenderer.h
    │   ├── stm32BspFrameBuffer.xml
    │   ├── tcmenu-plugin.xml
    │   ├── tfteSpi
    │   │   ├── tcMenuTfteSpi.cpp
    │   │   └── tcMenuTfteSpi.h
    │   ├── tftSpiDisplay.xml
    │   ├── u8g2Display.xml
    │   ├── u8g2Driver
    │   │   ├── tcMenuU8g2.cpp
    │   │   └── tcMenuU8g2.h
    │   ├── u8g2SimpleBuilder.xml
    │   ├── unoLcd
    │   │   ├── tcMenuLiquidCrystal.cpp
    │   │   └── tcMenuLiquidCrystal.h
    │   └── upDownSwitchesInput.xml
    ├── core-remote
    │   ├── embeddedJetty
    │   │   ├── TcJettyWebServer.java
    │   │   └── TcJettyWebSocketEndpoint.java
    │   ├── embedJSArchive
    │   │   └── embed-control-js-designer-latest.zip
    │   ├── ethernetSrc
    │   │   ├── EthernetTransport.cpp
    │   │   └── EthernetTransport.h
    │   ├── Images
    │   │   ├── embedCONTROLJS.png
    │   │   ├── embedCONTROL.png
    │   │   ├── esp8266.jpg
    │   │   ├── ethernet-shield.jpg
    │   │   ├── mbed-ethernet.jpg
    │   │   ├── menu-in-menu.png
    │   │   ├── no-display.png
    │   │   ├── serial-connection.jpg
    │   │   └── simhub.jpg
    │   ├── JavaNetworkRemote.xml
    │   ├── JavaWebServerRemote.xml
    │   ├── mbedEthernet
    │   │   ├── MBedEthernetTransport.cpp
    │   │   └── MBedEthernetTransport.h
    │   ├── noRemote.xml
    │   ├── remoteEthernet.xml
    │   ├── remoteMbedEthernet.xml
    │   ├── remoteSerialESP.xml
    │   ├── remoteSerial.xml
    │   ├── remoteWifiESP.xml
    │   ├── serialSrc
    │   │   ├── SerialTransport.cpp
    │   │   └── SerialTransport.h
    │   ├── simhub
    │   │   ├── SimhubConnector.cpp
    │   │   └── SimhubConnector.h
    │   ├── simhubConnector.xml
    │   └── tcmenu-plugin.xml
    └── core-themes
        ├── coolBlue
        │   ├── ThemeCoolBlueModern.h
        │   └── ThemeCoolBlueTraditional.h
        ├── CoolBlueModern.xml
        ├── CoolBlueTraditional.xml
        ├── darkMode
        │   ├── ThemeDarkModeModern.h
        │   └── ThemeDarkModeTraditional.h
        ├── DarkModeModern.xml
        ├── DarkModeTraditional.xml
        ├── Images
        │   ├── no-theme.png
        │   ├── theme-cool-blue-modern.jpg
        │   ├── theme-cool-blue-traditional.jpg
        │   ├── theme-dark-mode-modern.jpg
        │   ├── theme-dark-mode-traditional.jpg
        │   ├── theme-oled-bordered.jpg
        │   └── theme-oled-inverse.jpg
        ├── ManualTheme.xml
        ├── MonoBorderedTitle.xml
        ├── MonoInverseTitle.xml
        ├── monoThemes
        │   ├── ThemeMonoBordered.h
        │   └── ThemeMonoInverse.h
        ├── NoThemeSelected.xml
        └── tcmenu-plugin.xml

28 directories, 115 files
```
[tcmenu.log.0.txt](https://github.com/davetcc/tcMenu/files/10288264/tcmenu.log.0.txt)

davetcc, 2022-12-22:
Yes, that looks absolutely correct, when you're in the code generator dialog, what happens when you try to click on the image button highlighted below for the display plugin?

[image]

There's no exceptions or errors in the log, so if something is going wrong, it for some reason is not getting logged.

davetcc, 2022-12-22:
For the libraries need updating warning, if you click on that it will take you to general settings where you can choose either to set an Arduino and Libraries directory, or tell the designer that you're using platformio/other build system and don't have libraries. To be honest in the learning period, it's often useful to have the libraries somewhere local just for the packaged examples.

khvalera, 2022-12-22:
I'm sorry, I didn't think to click on the image :) It works, I will try. Can I package the AUR archlinux if you don't mind?

davetcc, 2022-12-28:
Apologies for the delay, this happened through the Chrismas holidays in the UK. In theory I have no problem with it going into that repo, and in fact it would be the best approach. The only slight concerns I have are around testing it sufficiently to ensure it is a good build, and maintaining it going forward. 

As far as I know, none of the other committers and contributors have the ability to package this to that repo. So unless someone does the release each time, it could easily get behind. There are not that many releases, generally less than one a month. 

My other concern is that we used to have the designer on both Windows and macOS stores, but people complained (sometimes bitterly) about it being upgraded automatically. These days the designer is a lot more stable, but even so, there are sometimes changes that require a round trip through the code generator, and not everyone wants to take those immediately.