TcMenu/tcMenu #273: Headers not included

rosenhauer, 2023-01-03:
Error when using Touch input:

`In file included from XXX_menu.cpp:12:0:
XXX_menu.h:31:8: error: 'MenuTouchScreenManager' does not name a type
 extern MenuTouchScreenManager touchScreen;
`
In v3.0.0 the `#include <graphics/MenuTouchScreenEncoder.h>` line is not included in the XXX_menu.h file when adding touch interface to the menu.  

After adding it to line 18 the menu compiles.

davetcc, 2023-01-04:
Thanks I’ll take a look at this when I get some time and feed back.

davetcc, 2023-01-04:
In fact out of the 4 boards one of them has different rotations. The resistive touch screen IIRC. Also you could see a situation where the touch overlay was applied wrongly on a batch and needed inverting.

davetcc, 2023-01-09:
As per the other comment, I'm doing some work on touch at the moment to support a couple of new interfaces, and tidy up the existing support. I'll check this at the same time.

davetcc, 2023-02-12:
I think I have all the cases in 3.1, if not after upgrading please do reopen and I'll look again.

closed 2023-02-12 by davetcc