TcMenu/tcMenu #275: Unable to compile with 3.0.0 version (using platformIO)

luwenjin, 2023-01-05:
I've updated the libs and regenerated the code, but got compile errors.

Error Messages
-------
```
Scanning dependencies...
Dependency Graph
|-- tcMenu @ 3.0.0
|   |-- IoAbstraction @ 3.0.2
|   |   |-- SimpleCollections @ 1.2.3
|   |   |-- TaskManagerIO @ 1.3.5
|   |-- tcUnicodeHelper @ 0.1.0
|   |   |-- IoAbstraction @ 3.0.2
|   |   |   |-- SimpleCollections @ 1.2.3
|   |   |   |-- TaskManagerIO @ 1.3.5
|   |   |-- U8g2 @ 2.34.5
|   |-- SimpleCollections @ 1.2.3
|-- IoAbstraction @ 3.0.2
|   |-- SimpleCollections @ 1.2.3
|   |-- TaskManagerIO @ 1.3.5
|-- tcUnicodeHelper @ 0.1.0
|   |-- IoAbstraction @ 3.0.2
|   |   |-- SimpleCollections @ 1.2.3
|   |   |-- TaskManagerIO @ 1.3.5
|   |-- U8g2 @ 2.34.5
|-- U8g2 @ 2.34.5
|-- FastLED @ 3.5.0
|-- ArduinoJson @ 6.19.4
|-- Nanopb @ 0.4.6+4
|-- rBase64 @ 1.1.1
|-- TaskManagerIO @ 1.3.5
Building in release mode
Compiling .pio/build/slave/liba7c/tcUnicodeHelper/tcUnicodeHelper.cpp.o
Compiling .pio/build/slave/lib4d7/tcMenu/extras/DrawableDashboard.cpp.o
.pio/libdeps/slave/tcUnicodeHelper/src/tcUnicodeHelper.cpp:8: error: ignoring #pragma clang diagnostic [-Werror=unknown-pragmas]
 #pragma clang diagnostic push
 
.pio/libdeps/slave/tcUnicodeHelper/src/tcUnicodeHelper.cpp:9: error: ignoring #pragma ide diagnostic [-Werror=unknown-pragmas]
 #pragma ide diagnostic ignored "UnreachableCode"
 
.pio/libdeps/slave/tcUnicodeHelper/src/tcUnicodeHelper.cpp:207: error: ignoring #pragma clang diagnostic [-Werror=unknown-pragmas]
 #pragma clang diagnostic pop
 
In file included from .pio/libdeps/slave/tcMenu/src/extras/DrawableDashboard.cpp:2:
.pio/libdeps/slave/tcMenu/src/extras/DrawableDashboard.h: In constructor 'DashMenuItem::DashMenuItem(MenuItem*, tcgfx::Coord, DashDrawParameters*, int, const char*, int)':
.pio/libdeps/slave/tcMenu/src/extras/DrawableDashboard.h:218:9: error: 'DashMenuItem::valueWidth' will be initialized after [-Werror=reorder]
     int valueWidth;
         ^~~~~~~~~~
.pio/libdeps/slave/tcMenu/src/extras/DrawableDashboard.h:216:11: error:   'tcgfx::Coord DashMenuItem::titleExtents' [-Werror=reorder]
     Coord titleExtents;
           ^~~~~~~~~~~~
.pio/libdeps/slave/tcMenu/src/extras/DrawableDashboard.cpp:153:1: error:   when initialized here [-Werror=reorder]
 DashMenuItem::DashMenuItem(MenuItem *theItem, Coord topLeft, DashDrawParameters* params, int numCharsInValue,
 ^~~~~~~~~~~~
cc1plus: some warnings being treated as errors
*** [.pio/build/slave/liba7c/tcUnicodeHelper/tcUnicodeHelper.cpp.o] Error 1
cc1plus: some warnings being treated as errors
*** [.pio/build/slave/lib4d7/tcMenu/extras/DrawableDashboard.cpp.o] Error 1
================================================ [FAILED] Took 20.50 seconds ================================================

Environment    Status    Duration
-------------  --------  ------------
slave          FAILED    00:00:20.503
```

davetcc, 2023-01-05:
Certainly is, it should always compile under strict settings.

davetcc, 2023-01-05:
You indirectly created the first issue in tcUnicode :)

https://github.com/davetcc/tcUnicodeHelper/issues/1

Also 

https://github.com/davetcc/tcMenuLib/issues/174

closed 2023-01-06 by davetcc