TcMenu/tcMenu #276: Font offset not calculated correctly when using fonts.

rosenhauer, 2023-01-05:
When using non built in fonts (tested with Adafruit fonts) the offset is incorrectly calculated.

In tcAdaFruitGfx.cpp internalTextExtents function the line:
`bl = glyphHeight + pgm_read_byte(&allGlyphs[glIdx].yOffset);`
should be 
`bl = glyphHeight + (signed char)pgm_read_byte(&allGlyphs[glIdx].yOffset);`
so that negative offsets are calculated correctly.  But this still just uses the offset of the last character tested - the left parentheses.  

or preferably it can be changed  by eliminating the bl calculation and replacing 

`if(baseline) *baseline = bl;
`
with 
`if(baseline) *baseline = font->yAdvance - height;`

This uses the line spacing for the font minus the tallest character.

davetcc, 2023-01-08:
Thanks for pointing this out, I look at this and try it out for both tcUnicode and native fonts. They both use the same calculation. I have a graphical example that shows this and can try it out.

davetcc, 2023-01-24:
Many thanks, this somehow escaped through testing, I have fixed this as per your suggestion and it will be in 3.1 which is due very soon.

davetcc, 2023-01-24:
I had not understood how serious this was, or I would have removed the touch change from 3.1 and released it earlier. As it is now 3.1 is probably a few days from release so it will go into that.

davetcc, 2023-01-24:
The fix is already applied in - https://github.com/davetcc/tcMenu/blob/master/xmlPlugins/core-display/adaGfxDriver/tcMenuAdaFruitGfx.cpp

rosenhauer, 2023-01-24:
You're welcome!  Always glad to help out if I can.

closed 2023-02-12 by davetcc