TcMenu/tcMenu #277: Are there any 'activeItemHasChanged' like method for RuntimeMenuItem?

luwenjin, 2023-01-06:
Can I get notified when I select different items under a RuntimeMenuItem?
I've tried `MenuManagerObserver.activeItemHasChanged`, but haven't captured that event.

davetcc, 2023-01-07:
We can't do that, there is only one menu item associated with them. The way that runtime items work is that you implement the callback that receives all the different events, you can get this from the requested row in the callback. You can read about these in the following guide - https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/based-on-runtimemenuitem/

In either 3.1 or 3.2 there's a plan to make it so that you can override the callback for all types of runtime item, not just lists and scroll choices, meaning you'd have far more flexibility to handle all these cases.

luwenjin, 2023-01-07:
How about adding a new RenderFnMode to 
```
int CALLBACK_FUNCTION fnAdditionalCountListRtCall(RuntimeMenuItem * item, uint8_t row, 
                                                  RenderFnMode mode, 
                                                  char * buffer, int bufferSize) {
```
maybe `RENDERFN_ACTIVATE` ?

davetcc, 2023-01-07:
https://github.com/davetcc/tcMenuLib/issues/178 finished

closed 2023-01-07 by davetcc