TcMenu/tcMenu #278: Suggestion: get/set the whole navigation history

luwenjin, 2023-01-06:
I want to put my app into deep sleep after long inactivity, and restore the UI after any key hit.
The deep sleep of ESP32 is almost the same as power off, and can not hold any complicated state of the app.

So I'd like to have access to the navigation history data, and encode/save it to EEPROM.
When wake up, I would decode the navigation history from EEPROM, restore the navigation history and display the previous UI.

I'd like to suggest 2 API in the MenuManager class:
1. get the navigation history, return a datatype that is easy to encode.
2. set the navigation history and restore the menu by history.

davetcc, 2023-01-07:
I think it would be possible to provide a function that could retrieve the depth (number of entries in the nav array) and each entry by index. That way you could persist it easily enough and it would have minimal overhead on others. Putting it back similarly would need an accessor function that allowed you to set each level of the navigation. 

These are menu item pointers though, so you'd need to somehow de-reference these into IDs and then store the IDs, there are many functions in https://www.thecoderscorner.com/ref-docs/tcmenu/html/_menu_iterator_8h.html that may help out there.

I would suggest something like this to get the navigation, you could combine this with getting the information from menuMgr itself and save the lot:

```
auto nav= menuMgr.getNavigationStore();
for(int i=0;i<nav.getNavigtionDepth(); i++) {
    yourSaveToEeprom(nav.getActiveItemAt(i), nav.getSubItemAt(i), nav.getRootItemAt(i));
}

Then to reconstruct on restart, first do `nav.clear()`

and on the way back, for each entry retrieved, it may be safer to actually do the navigation through menuMgr.

But, are you sure this is a good idea coming out of deep sleep deep into the menu structure, the device may have been asleep for some time, and it may be a different user that wakes it up, in an unexpected area of the menu.

Do you have any secure sub-menus? If so you can't actually replay the navigation, but maybe that's a good thing in that case as otherwise another user could bypass the pin security.

luwenjin, 2023-01-07:
I think the way your suggest is good enough for me, I can achieve what I want to do.

"Wake up and restore the UI before sleep" is pretty standard interaction pattern, mac works this way, so do mobile phones.
They work so smoothly, that user don't even realize it's some sort of hibernation and waking up.
That's what I want to achieve on my device, users can leave the device aside, after a period of time, it goes to sleep and save power. When the user want to use the device again, just press any key to wake it up and continue the work.